AxisCommunications/acap3-examples #34: Error in larod inference request

waltermaffy, 2020-09-16:
Hello, 
I'm trying to build an acap for object detection to run on an axis device with a TPU.
I'm adapting the app vdo-larod by changing the larod model with an obj-det one (larod converted), and it connect successfully to the stream video. 
However, I'm not able to create an inference request. I get "Failed creating inference request: Input tensors are invalid: Tensor 1 is null", do you know there this error could come from?

carlcn, 2020-09-18:
This error most likely appears from following line in file vdo_larod.c.

333    infReq = larodCreateInferenceRequest(model, inputTensors, 1, outputTensors, 1, &error);

Parameter inputTensors has one tensor, but the third parameter "1" has been changed to a larger number.
Please check that the values from following line is used in the call above:
300    inputTensors = larodCreateModelInputs(model, &numInputs, &error);

hussanm, 2020-09-18:
> Hello,
> I'm trying to build an acap for object detection to run on an axis device with a TPU.
> I'm adapting the app vdo-larod by changing the larod model with an obj-det one (larod converted), and it [...]

Could you please provide the following details related to the problem.

- What Axis camera model and firmware version are you running the example on?
- Stack trace error

waltermaffy, 2020-09-18:
Thank you for your help.

The camera is an AXIS Q1615 Mk III with firmware: 10.0_beta15

I don't know where to find the stack trace error since I don't get any warning or error during compilation but only an error in the log of the application. 
This is what i get 

> 2020-09-18T11:56:31.641+02:00 axis-accc8efc919e [ INFO    ] vdo_larod[0]: starting vdo_larod
2020-09-18T11:56:31.675+02:00 axis-accc8efc919e [ INFO    ] vdo_larod[14035]: Starting ...
2020-09-18T11:56:31.701+02:00 axis-accc8efc919e [ INFO    ] vdo_larod[14035]: Creating VDO image provider and creating stream 1920 x 1080
2020-09-18T11:56:31.701+02:00 axis-accc8efc919e [ INFO    ] vdo_larod[14035]: Dump of vdo stream settings map =====
2020-09-18T11:56:31.701+02:00 axis-accc8efc919e [ INFO    ] vdo_larod[14035]: chooseStreamResolution: We select stream w/h=1920 x 1080 based on VDO channel info.
2020-09-18T11:56:31.818+02:00 axis-accc8efc919e [ INFO    ] vdo_larod[14035]: Setting up larod connection with chip 4 and model /usr/local/packages/vdo_larod/model/ssdlite_mobiledet_edgetpu_320/uint8/model_edgetpu.larod
2020-09-18T11:56:32.034+02:00 axis-accc8efc919e [ ERR     ] vdo_larod[14035]: Model has 12 outputs, app only supports 1 output tensor.
2020-09-18T11:56:32.034+02:00 axis-accc8efc919e [ INFO    ] vdo_larod[14035]: Creating temporary files and memmaps for inference input and output tensors
2020-09-18T11:56:32.034+02:00 axis-accc8efc919e [ INFO    ] vdo_larod[14035]: createAndMapTmpFile: Setting up a temp fd with pattern /tmp/larod.in.test-XXXXXX and size 6220800
2020-09-18T11:56:32.034+02:00 axis-accc8efc919e [ INFO    ] vdo_larod[14035]: createAndMapTmpFile: Setting up a temp fd with pattern /tmp/larod.out.test-XXXXXX and size 2048

By commenting the check on (numOutputs != 1) and changing this line of code 
`infReq = larodCreateInferenceRequest(model, inputTensors, 1, outputTensors, 12, &error);`

I get : 

> 2020-09-18T12:06:51.159+02:00 axis-accc8efc919e [ INFO    ] vdo_larod[15321]: Starting ...
2020-09-18T12:06:51.176+02:00 axis-accc8efc919e [ INFO    ] vdo_larod[15321]: Creating VDO image provider and creating stream 1920 x 1080
2020-09-18T12:06:51.176+02:00 axis-accc8efc919e [ INFO    ] vdo_larod[15321]: Dump of vdo stream settings map =====
2020-09-18T12:06:51.176+02:00 axis-accc8efc919e [ INFO    ] vdo_larod[15321]: chooseStreamResolution: We select stream w/h=1920 x 1080 based on VDO channel info.
2020-09-18T12:06:51.324+02:00 axis-accc8efc919e [ INFO    ] vdo_larod[15321]: Setting up larod connection with chip 4 and model /usr/local/packages/vdo_larod/model/ssdlite_mobiledet_edgetpu_320/uint8/model_edgetpu.larod
2020-09-18T12:06:51.666+02:00 axis-accc8efc919e [ INFO    ] vdo_larod[15321]: Creating temporary files and memmaps for inference input and output tensors
2020-09-18T12:06:51.666+02:00 axis-accc8efc919e [ INFO    ] vdo_larod[15321]: Start fetching video frames from VDO
2020-09-18T12:06:51.666+02:00 axis-accc8efc919e [ INFO    ] vdo_larod[15321]: createAndMapTmpFile: Setting up a temp fd with pattern /tmp/larod.in.test-XXXXXX and size 6220800
2020-09-18T12:06:51.666+02:00 axis-accc8efc919e [ INFO    ] vdo_larod[15321]: createAndMapTmpFile: Setting up a temp fd with pattern /tmp/larod.out.test-XXXXXX and size 2048
2020-09-18T12:06:51.886+02:00 axis-accc8efc919e [ ERR     ] vdo_larod[15321]: Unable to run inference on model /usr/local/packages/vdo_larod/model/ssdlite_mobiledet_edgetpu_320/uint8/model_edgetpu.larod: Could not append tensor fd details: Invalid argument (22)
2020-09-18T12:06:51.886+02:00 axis-accc8efc919e [ INFO    ] vdo_larod[15321]: Converted image in 214 ms

hussanm, 2020-09-18:
Please note that the application has a limitation and it supports 1 input and output tensor for now. However, the trace shows that your model has 12 outputs. Could you also share the model with us to reproduce the error?

waltermaffy, 2020-09-18:
Dear @hussanm 
I'm trying to build an object detector on top of larod, so a tensorOutput > 1 is required. I did not find any docs on larod expect larod.h so building on top of that without any other resources results difficult. I know that AxisObjectAnalytics developers team managed to do that so I don't think it's impossible. 

I attach the zipped app with Dockerfile and Makefile that loads and convert the model. 
[vdo-larod.zip](https://github.com/AxisCommunications/acap3-examples/files/5245148/vdo-larod.zip)

hussanm, 2020-09-21:
Thank you for your feedback. This example is designed to support the single input and output sensor for the sake of simplicity and allow the developers to extent it based on their own needs. Therefore, commenting on the check on (numOutputs != 1) would not be enough to execute the example to support multiple tensors. You will need to tweak code more than that to make it work for multiple tensors as is the case with your code (guessing your code here). Essentially, you would need to modify the code below to support multiple tensors and remove the part that prints results since it relies on only using one output tensor. You need to loop over the larodSetTensorFd for as many tensors as the model requires, instead of just using it on one tensor and in calling larodCreateInferenceRequest you need to replace the 1s with the number of inputs and outputs they have. 

There are two options to fix it. First, you can make the changes as suggested above or wait for the updated example supporting multiple tensors as we will address this feature request. 
 

   inputTensors = larodCreateModelInputs(model, &numInputs, &error);
    if (!inputTensors) {
        syslog(LOG_ERR, "Failed retrieving input tensors: %s", error->msg);
        goto end;
    }
    // This app only supports 1 input tensor right now.
    if (numInputs != 1) {
        syslog(LOG_ERR, "Model has %zu inputs, app only supports 1 input tensor.",
                 numInputs);
        goto end;
    }
    if (!larodSetTensorFd(inputTensors[0], larodInputFd, &error)) {
        syslog(LOG_ERR, "Failed setting input tensor fd: %s", error->msg);
        goto end;
    }

    outputTensors = larodCreateModelOutputs(model, &numOutputs, &error);
    if (!outputTensors) {
        syslog(LOG_ERR, "Failed retrieving output tensors: %s", error->msg);
        goto end;
    }
    // This app only supports 1 output tensor right now.
    if (numOutputs != 1) {
        syslog(LOG_ERR, "Model has %zu outputs, app only supports 1 output tensor.",
                 numOutputs);
        goto end;
    }
    if (!larodSetTensorFd(outputTensors[0], larodOutputFd, &error)) {
        syslog(LOG_ERR, "Failed setting output tensor fd: %s", error->msg);
        goto end;
    }

waltermaffy, 2020-09-23:
Dear @hussanm, 
Thank your for your reply. What do you mean with "loop over the larodSetTensorFd" ? 
Do you have any estimated time on when an example acap using larod for object detection will be published?

hussanm, 2020-09-29:
We will publish a new example in 2-3 weeks which will cover the following parts.
- Multiple tensors support
- How to train a model and run it on the camera
- How to run a pre-trained model on the camera

hussanm, 2020-10-14:
The new example is available 
[https://github.com/AxisCommunications/acap3-examples/tree/master/tensorflow-to-larod](url)

closed 2020-10-14 by hussanm