AxisCommunications/acap3-examples #37: vdoencodeclient output file is corrupt

mattfeury, 2020-09-29:
hello,

i am trying to test the vdoencodeclient application. i'm able to successfully build and install it on the device. via ssh, i am able to invoke it via something like `./vdoencodeclient --format h264 --frames 60 --output wtf.mp4`. however, the output file appears to be invalid. it is not able to be played via vlc (attempted multiple extensions – .mp4, .avc, .h264. the latter actually gave me something with tons of artifacts but had at least something visible). 

jpeg format successfully gives me a single frame, whereas the h265 format errors out immediately with:

```
2020-09-29T00:04:00.343-04:00 axis-accc8ef3c0d4 [ INFO    ] vdoencodeclient[8143]: vdo-encode-client: DBus call failed: Message recipient disconnected from message bus without replying
```

below is an example of a log with an example file:

```
2020-09-28T23:55:42.726-04:00 axis-accc8ef3c0d4 [ INFO    ] vdoencodeclient[7464]: Starting stream: h264, 1024x768, 5 fps
2020-09-28T23:55:42.787-04:00 axis-accc8ef3c0d4 [ INFO    ] vdoencodeclient[7464]: frame =    0, type = I, size = 26617
2020-09-28T23:55:42.986-04:00 axis-accc8ef3c0d4 [ INFO    ] vdoencodeclient[7464]: frame =    1, type = P, size = 854
2020-09-28T23:55:43.186-04:00 axis-accc8ef3c0d4 [ INFO    ] vdoencodeclient[7464]: frame =    2, type = P, size = 2645
2020-09-28T23:55:43.386-04:00 axis-accc8ef3c0d4 [ INFO    ] vdoencodeclient[7464]: frame =    3, type = P, size = 3115
2020-09-28T23:55:43.586-04:00 axis-accc8ef3c0d4 [ INFO    ] vdoencodeclient[7464]: frame =    4, type = P, size = 1153
2020-09-28T23:55:43.787-04:00 axis-accc8ef3c0d4 [ INFO    ] vdoencodeclient[7464]: frame =    5, type = P, size = 1986
2020-09-28T23:55:43.986-04:00 axis-accc8ef3c0d4 [ INFO    ] vdoencodeclient[7464]: frame =    6, type = P, size = 578
2020-09-28T23:55:44.186-04:00 axis-accc8ef3c0d4 [ INFO    ] vdoencodeclient[7464]: frame =    7, type = P, size = 935
2020-09-28T23:55:44.386-04:00 axis-accc8ef3c0d4 [ INFO    ] vdoencodeclient[7464]: frame =    8, type = P, size = 1153
2020-09-28T23:55:44.586-04:00 axis-accc8ef3c0d4 [ INFO    ] vdoencodeclient[7464]: frame =    9, type = P, size = 609
2020-09-28T23:55:44.787-04:00 axis-accc8ef3c0d4 [ INFO    ] vdoencodeclient[7464]: frame =   10, type = P, size = 2808
2020-09-28T23:55:44.986-04:00 axis-accc8ef3c0d4 [ INFO    ] vdoencodeclient[7464]: frame =   11, type = P, size = 479
2020-09-28T23:55:45.186-04:00 axis-accc8ef3c0d4 [ INFO    ] vdoencodeclient[7464]: frame =   12, type = P, size = 648
2020-09-28T23:55:45.386-04:00 axis-accc8ef3c0d4 [ INFO    ] vdoencodeclient[7464]: frame =   13, type = P, size = 667
2020-09-28T23:55:45.585-04:00 axis-accc8ef3c0d4 [ INFO    ] vdoencodeclient[7464]: frame =   14, type = P, size = 662
2020-09-28T23:55:45.786-04:00 axis-accc8ef3c0d4 [ INFO    ] vdoencodeclient[7464]: frame =   15, type = P, size = 1872
2020-09-28T23:55:45.986-04:00 axis-accc8ef3c0d4 [ INFO    ] vdoencodeclient[7464]: frame =   16, type = P, size = 777
2020-09-28T23:55:46.186-04:00 axis-accc8ef3c0d4 [ INFO    ] vdoencodeclient[7464]: frame =   17, type = P, size = 1147
2020-09-28T23:55:46.386-04:00 axis-accc8ef3c0d4 [ INFO    ] vdoencodeclient[7464]: frame =   18, type = P, size = 2772
2020-09-28T23:55:46.586-04:00 axis-accc8ef3c0d4 [ INFO    ] vdoencodeclient[7464]: frame =   19, type = P, size = 1368
2020-09-28T23:55:46.786-04:00 axis-accc8ef3c0d4 [ INFO    ] vdoencodeclient[7464]: frame =   20, type = P, size = 2885
2020-09-28T23:55:46.987-04:00 axis-accc8ef3c0d4 [ INFO    ] vdoencodeclient[7464]: frame =   21, type = P, size = 520
2020-09-28T23:55:47.187-04:00 axis-accc8ef3c0d4 [ INFO    ] vdoencodeclient[7464]: frame =   22, type = P, size = 613
2020-09-28T23:55:47.386-04:00 axis-accc8ef3c0d4 [ INFO    ] vdoencodeclient[7464]: frame =   23, type = P, size = 1083
2020-09-28T23:55:47.585-04:00 axis-accc8ef3c0d4 [ INFO    ] vdoencodeclient[7464]: frame =   24, type = P, size = 1869
2020-09-28T23:55:47.786-04:00 axis-accc8ef3c0d4 [ INFO    ] vdoencodeclient[7464]: frame =   25, type = P, size = 1522
2020-09-28T23:55:47.987-04:00 axis-accc8ef3c0d4 [ INFO    ] vdoencodeclient[7464]: frame =   26, type = P, size = 614
2020-09-28T23:55:48.186-04:00 axis-accc8ef3c0d4 [ INFO    ] vdoencodeclient[7464]: frame =   27, type = P, size = 1023
2020-09-28T23:55:48.386-04:00 axis-accc8ef3c0d4 [ INFO    ] vdoencodeclient[7464]: frame =   28, type = P, size = 1588
2020-09-28T23:55:48.586-04:00 axis-accc8ef3c0d4 [ INFO    ] vdoencodeclient[7464]: frame =   29, type = P, size = 994
2020-09-28T23:55:48.786-04:00 axis-accc8ef3c0d4 [ INFO    ] vdoencodeclient[7464]: frame =   30, type = I, size = 26176
2020-09-28T23:55:48.986-04:00 axis-accc8ef3c0d4 [ INFO    ] vdoencodeclient[7464]: frame =   31, type = P, size = 896
2020-09-28T23:55:49.186-04:00 axis-accc8ef3c0d4 [ INFO    ] vdoencodeclient[7464]: frame =   32, type = P, size = 893
2020-09-28T23:55:49.386-04:00 axis-accc8ef3c0d4 [ INFO    ] vdoencodeclient[7464]: frame =   33, type = P, size = 860
2020-09-28T23:55:49.586-04:00 axis-accc8ef3c0d4 [ INFO    ] vdoencodeclient[7464]: frame =   34, type = P, size = 845
2020-09-28T23:55:49.786-04:00 axis-accc8ef3c0d4 [ INFO    ] vdoencodeclient[7464]: frame =   35, type = P, size = 2049
2020-09-28T23:55:49.986-04:00 axis-accc8ef3c0d4 [ INFO    ] vdoencodeclient[7464]: frame =   36, type = P, size = 664
2020-09-28T23:55:50.187-04:00 axis-accc8ef3c0d4 [ INFO    ] vdoencodeclient[7464]: frame =   37, type = P, size = 821
2020-09-28T23:55:50.386-04:00 axis-accc8ef3c0d4 [ INFO    ] vdoencodeclient[7464]: frame =   38, type = P, size = 929
2020-09-28T23:55:50.586-04:00 axis-accc8ef3c0d4 [ INFO    ] vdoencodeclient[7464]: frame =   39, type = P, size = 841
2020-09-28T23:55:50.786-04:00 axis-accc8ef3c0d4 [ INFO    ] vdoencodeclient[7464]: frame =   40, type = P, size = 3311
2020-09-28T23:55:50.986-04:00 axis-accc8ef3c0d4 [ INFO    ] vdoencodeclient[7464]: frame =   41, type = P, size = 506
2020-09-28T23:55:51.186-04:00 axis-accc8ef3c0d4 [ INFO    ] vdoencodeclient[7464]: frame =   42, type = P, size = 654
2020-09-28T23:55:51.387-04:00 axis-accc8ef3c0d4 [ INFO    ] vdoencodeclient[7464]: frame =   43, type = P, size = 738
2020-09-28T23:55:51.586-04:00 axis-accc8ef3c0d4 [ INFO    ] vdoencodeclient[7464]: frame =   44, type = P, size = 855
2020-09-28T23:55:51.786-04:00 axis-accc8ef3c0d4 [ INFO    ] vdoencodeclient[7464]: frame =   45, type = P, size = 2024
2020-09-28T23:55:51.986-04:00 axis-accc8ef3c0d4 [ INFO    ] vdoencodeclient[7464]: frame =   46, type = P, size = 584
2020-09-28T23:55:52.187-04:00 axis-accc8ef3c0d4 [ INFO    ] vdoencodeclient[7464]: frame =   47, type = P, size = 857
2020-09-28T23:55:52.387-04:00 axis-accc8ef3c0d4 [ INFO    ] vdoencodeclient[7464]: frame =   48, type = P, size = 864
2020-09-28T23:55:52.586-04:00 axis-accc8ef3c0d4 [ INFO    ] vdoencodeclient[7464]: frame =   49, type = P, size = 854
2020-09-28T23:55:52.786-04:00 axis-accc8ef3c0d4 [ INFO    ] vdoencodeclient[7464]: frame =   50, type = P, size = 3295
2020-09-28T23:55:52.985-04:00 axis-accc8ef3c0d4 [ INFO    ] vdoencodeclient[7464]: frame =   51, type = P, size = 546
2020-09-28T23:55:53.186-04:00 axis-accc8ef3c0d4 [ INFO    ] vdoencodeclient[7464]: frame =   52, type = P, size = 656
2020-09-28T23:55:53.386-04:00 axis-accc8ef3c0d4 [ INFO    ] vdoencodeclient[7464]: frame =   53, type = P, size = 773
2020-09-28T23:55:53.586-04:00 axis-accc8ef3c0d4 [ INFO    ] vdoencodeclient[7464]: frame =   54, type = P, size = 790
2020-09-28T23:55:53.786-04:00 axis-accc8ef3c0d4 [ INFO    ] vdoencodeclient[7464]: frame =   55, type = P, size = 3497
2020-09-28T23:55:53.987-04:00 axis-accc8ef3c0d4 [ INFO    ] vdoencodeclient[7464]: frame =   56, type = P, size = 10419
2020-09-28T23:55:54.186-04:00 axis-accc8ef3c0d4 [ INFO    ] vdoencodeclient[7464]: frame =   57, type = P, size = 6182
2020-09-28T23:55:54.386-04:00 axis-accc8ef3c0d4 [ INFO    ] vdoencodeclient[7464]: frame =   58, type = P, size = 6215
2020-09-28T23:55:54.587-04:00 axis-accc8ef3c0d4 [ INFO    ] vdoencodeclient[7464]: frame =   59, type = P, size = 14005
```

video is available here: https://drive.google.com/file/d/1vszkr6_fOMGETjtfK-5aGa-6rFgLBQ5b/view?usp=sharing

appreciate any feedback. thanks!

petterwa, 2020-10-02:
We suspect you might be testing on a camera model where vdo doesn't support h265. There are also a known issue with h264 on some camera models/firmware combinations.
Which camera model and firmware version are you testing on?

mattfeury, 2020-10-02:
yea i can confirm that h265 isn't supported on my device, so not totally surprised there. but def surprised to see that h264 is having trouble when i can easily make h264 videos using the recording api, e.g..

camera model: AXIS Q1786-LE
firmware: 9.80.2.2

petterwa, 2020-10-06:
I think you have same issue with h264 on this product  as VDO developers is already working on and they expect to have a solution for this issue in next firmware release - 10.2.

mattfeury, 2020-10-06:
ah i see. if this is a firmware issue, is there a recommend way for capturing video via the sdk? i would like to implement some preroll and basic filtering and support a good amount of devices. should i switch to the mediacapture sdk? just wondering a good recommended path forward if vdo is buggy. even if it's local executables or vapix apis.

petterwa, 2020-10-08:
VDO is the recommended API for capturing video from an ACAP 3 application.
There are alternatives such as the old capture api from ACAP 2 but it's not recommended since there are concerns with performance and the future support on new models.

petterwa, 2020-11-05:
Firmware 10.2 will soon be released with the fix in VDO.

closed 2020-11-05 by petterwa

mattfeury, 2020-11-06:
thanks! will this fix be backported to other firmware versions? it sounds like right now that libvdo will only function properly on 10.2 which will greatly limit its reach.

mattfeury, 2020-11-09:
sorry to double up, but just want to be clear. it's my understanding that the VDO SDK is not functional currently and will not be until 10.2 releases. is that correct? i am pretty interested in using it even if it means modifying my code in some way to get around a bug.

petterwa, 2020-11-10:
Yes, you need 10.2 to get h264 funtional from the vdo API on your camera. I don't think you have to wait much longer for 10.2 to go public.. 
The fix for h264 is not yet backported to next LTS (9.80.3). Do you need use h264 on LTS?
To get arround the h264 bug in vdo you could use different format e.g. jpeg or use the old capture api as backup.

mattfeury, 2020-11-10:
that's very helpful, petterwa. fills in my knowledge gaps here. i will have to find another approach to get h264 since I want to build an app that is fairly universal across devices. but ideally it is backported for myself or others who want to use it across all devices. the artpec chip seems super powerful but right now there's not a great way to utilize it without this.