AxisCommunications/acap3-examples #42: multiple output example

waltermaffy, 2020-10-26:
Hello,
I'm trying to implement an object detector following your guide on "tensorflow-to-larod".
I managed to convert the model to .larod and I mapped the input and output tensors to file descriptors.
As reference, considering N outputs (in my model are 12): 

```
larodOutputsAddr` = (void **) malloc ( numOutputs * sizeof(void *));
    for(int i=0; i < numOutputs; i++){
        larodOutputsAddr[i] = MAP_FAILED;
        if (!createAndMapTmpFile(CONV_OUT_FILES[i], args.outputBytes, &larodOutputsAddr[i], &larodOutputsFd[i])) {
            goto end;
        }
    }
for(int i=0; i < numOutputs ; i++) {
        if (!larodSetTensorFd(outputTensors[i], larodOutputsFd[i], &error)) {
            syslog(LOG_ERR, "Failed setting output tensor fd: %s", error->msg);
            goto end;
        }
    }
```
The code above run correctly and I was able to run `larodRunInference( )` without any issues. 
Unfortunately, I'm having some troubles when I have to read the output tensors. I think its because each output tensor its not a single scalar but it may be an array or for example in python a dictionary. 
Do you have any advice on how to read multiple output tensors or would you provide an Object detection example acap?
Thank you ,
Walter

joha-axis, 2020-10-27:
Hi,

Could you please provide the following details related to the problem:

- What are the dimensions of your output tensors?

- What is the value of the args.outputBytes variable? This is provided as an argument to the application through the `package.conf` file's APPOPTS key

- How are you trying to read the output tensors?

waltermaffy, 2020-10-27:
Hi,
When I run `outputTensors = larodCreateModelOutputs(model, &numOutputs, &error);` 
numOutputs is **12**, but I didn't get what each of these output tensors represent. 
(The model I'm trying to run is this one: http://download.tensorflow.org/models/object_detection/ssdlite_mobiledet_edgetpu_320x320_coco_2020_05_19.tar.gz)
In order to know the output size of my tflite model I run these lines of code in python:
```
TFLITE_MODEL = "tflite_models/model.tflite"
tflite_interpreter = tf.lite.Interpreter(model_path=TFLITE_MODEL)
input_details = tflite_interpreter.get_input_details()
output_details = tflite_interpreter.get_output_details()
print("== Input details ==")
print("name:", input_details[0]['name'])
print("shape:", input_details[0]['shape'])
print("type:", input_details[0]['dtype'])
print("\n== Output details ==")
print("name:", output_details[0]['name'])
print("shape:", output_details[0]['shape'])
print("type:", output_details[0]['dtype']) 

for o in output_details:
  print(o)
```
And I get: 

```
== Input details ==
name: normalized_input_image_tensor
shape: [  1 320 320   3]
type: <class 'numpy.uint8'>

== Output details ==
name: BoxPredictor_0/BoxEncodingPredictor/BiasAdd
shape: [ 1 20 20 12]
type: <class 'numpy.uint8'>

{'name': 'BoxPredictor_0/BoxEncodingPredictor/BiasAdd', 'index': 0, 'shape': array([ 1, 20, 20, 12], dtype=int32), 'dtype': <class 'numpy.uint8'>, 'quantization': (0.0235294122248888, 0)}
{'name': 'BoxPredictor_0/ClassPredictor/BiasAdd', 'index': 6, 'shape': array([  1,  20,  20, 273], dtype=int32), 'dtype': <class 'numpy.uint8'>, 'quantization': (0.0235294122248888, 0)}
{'name': 'BoxPredictor_1/BoxEncodingPredictor/BiasAdd', 'index': 12, 'shape': array([ 1, 10, 10, 24], dtype=int32), 'dtype': <class 'numpy.uint8'>, 'quantization': (0.0235294122248888, 0)}
{'name': 'BoxPredictor_1/ClassPredictor/BiasAdd', 'index': 18, 'shape': array([  1,  10,  10, 546], dtype=int32), 'dtype': <class 'numpy.uint8'>, 'quantization': (0.0235294122248888, 0)}
{'name': 'BoxPredictor_2/BoxEncodingPredictor/BiasAdd', 'index': 24, 'shape': array([ 1,  5,  5, 24], dtype=int32), 'dtype': <class 'numpy.uint8'>, 'quantization': (0.0235294122248888, 0)}
{'name': 'BoxPredictor_2/ClassPredictor/BiasAdd', 'index': 30, 'shape': array([  1,   5,   5, 546], dtype=int32), 'dtype': <class 'numpy.uint8'>, 'quantization': (0.0235294122248888, 0)}
{'name': 'BoxPredictor_3/BoxEncodingPredictor/BiasAdd', 'index': 36, 'shape': array([ 1,  3,  3, 24], dtype=int32), 'dtype': <class 'numpy.uint8'>, 'quantization': (0.0235294122248888, 0)}
{'name': 'BoxPredictor_3/ClassPredictor/BiasAdd', 'index': 42, 'shape': array([  1,   3,   3, 546], dtype=int32), 'dtype': <class 'numpy.uint8'>, 'quantization': (0.0235294122248888, 0)}
{'name': 'BoxPredictor_4/BoxEncodingPredictor/BiasAdd', 'index': 48, 'shape': array([ 1,  2,  2, 24], dtype=int32), 'dtype': <class 'numpy.uint8'>, 'quantization': (0.0235294122248888, 0)}
{'name': 'BoxPredictor_4/ClassPredictor/BiasAdd', 'index': 54, 'shape': array([  1,   2,   2, 546], dtype=int32), 'dtype': <class 'numpy.uint8'>, 'quantization': (0.0235294122248888, 0)}
{'name': 'BoxPredictor_5/BoxEncodingPredictor/BiasAdd', 'index': 60, 'shape': array([ 1,  1,  1, 24], dtype=int32), 'dtype': <class 'numpy.uint8'>, 'quantization': (0.0235294122248888, 0)}
{'name': 'BoxPredictor_5/ClassPredictor/BiasAdd', 'index': 66, 'shape': array([  1,   1,   1, 546], dtype=int32), 'dtype': <class 'numpy.uint8'>, 'quantization': (0.0235294122248888, 0)}
```
In the file `package.conf` I set 2048 bytes for each output tensor but I'm not sure it's enough.
I just need to understand which one of these output represent the useful information that I need, ( for each detection [x1, y1, x2, y2, class, score]) and how to get them from the mapped memory address. 

I'm reading the output tensors using these lines of code 

```
for(int i = 0; i < numOutputs; i++ ){
            uint8_t * tmp = (uint8_t *) larodOutputsAddr[i]; 
            syslog(LOG_INFO, "%d) pointer at: %p", i, tmp);
            //iter tensor byte x byte 
            for (size_t j = 0; j < args.outputBytes ; j++) {
                syslog(LOG_INFO, "%.2f", (float) tmp[j]);
            }
 }
```

joha-axis, 2020-10-28:
Hi,

I do not recognize the output signature of your model. Could you try to use one of the object detection models provided by [coral.ai](https://coral.ai/models/)? For example the https://github.com/google-coral/edgetpu/raw/master/test_data/ssd_mobilenet_v2_coco_quant_postprocess_edgetpu.tflite. 

This model takes an (300, 300, 3) image as input and has four outputs, which follow the [TensorFlow object detection signature](https://www.tensorflow.org/lite/models/object_detection/overview#output_signature). Note that this model outputs float32, so you need to allocate 4 bytes per value and be aware of this when trying to read the output. With a maximum number of objects detected simulatenously of 20, you would need to allocate 4 * 4 * 20B, 4 * 20B, 4 * 20B, 4B to the four outputs respectively (i.e., replace the args.outputBytes value with this output-specific value).

waltermaffy, 2020-11-04:
Hi Joha,
Thank you for your answer. I managed to run successfully mobilenet-v2 detector using larod on our axis camera. 
Do you know what is the best way to send the detections through the network? 
I see that it is possible to send ONVIF Events, but is this the only way? 
Thanks

mikaelli-axis, 2020-11-05:
Can you describe where you want to send the dections to? The main purpose of the axevent API is to communicate events with systems that connectes to the device over the network using VAPIX or ONVIF, typically a VMS. If this is what you indent?

waltermaffy, 2020-11-05:
I need to send the detections over the network to a server that has to do some operations with them, not a Axis VMS.
I'd like to know if the only way to do that is to subscribe the server to the RTSP stream and Onvif event, or is it possible to use some others protocol sush as MQTT. 
Regarding Axis/Onvif Events, do you have some advice on how to use axevent API for this task? Do I need a secondary thread?

mikaelli-axis, 2020-11-05:
There is nothing preventing you from using your own preferred protocols for sending events or meta data, but at the moment you will have to add client libraries or other necessary tools as part of your application. That said, MQTT and support for other messaging standards have been requested before and your interest in these such features is noted. 

I think this discussion has grown beyond the initial issue with the example, so if you have further questions around this I would suggest you head on over to our developer pages on https://developer.axis.com and check out the documentation there.

waltermaffy, 2020-11-05:
Thank you very much!

closed 2020-11-05 by waltermaffy