AxisCommunications/acap3-examples #44: Are there any APIs to expose a custom RTSP stream using ACAP?

mdsingh2013, 2020-10-30:
I essentially am successfully able to use vdo_stream, are there any apis to expose these frames with some custom data out to a RTSP stream or something?

petterwa, 2020-11-03:
I'm checking with our experts on streaming what options there are for a custom RTSP stream from an ACAP.

johan-bjareholt, 2020-11-04:
There is unfortunately no API to create a custom RTSP stream with your own data from an ACAP.

What do you mean with custom data?
If you want an overlay you could make an overlay with AxOverlay without the need of a custom RTSP stream.

But if you mean custom data in any format inside the stream that will be trickier.
One way would be to create your own HTTP endpoint and then insert a SEI frame in the h26x stream from vdo, but that's a bit complicated to do.
A custom HTTP endpoint can be done with AxHTTP, you can make your own endpoint which upon request starts a stream from VDO and then proceed to send h264/h265 inside mp4/mkv in the response.
For many use-cases this might be enough, but not for all of course.
Such an HTTP endpoint could be used straight from a browser as it supports JPEG, MJPEG and h26x inside mp4 (not mkv however).

mdsingh2013, 2020-11-24:
I just wanted extremely accurate timestamps along with the frames. The vdo stream provides an easy way to do that, which I was not able to get from the existing RTSP stream, so was wondering how I can expose the frame + timestamp for consumption outside the camera?

johan-bjareholt, 2020-11-24:
@mdsingh2013 If you only want the timestamps you can start an ordinary RTSP stream and enable userdata for h264/mpeg. How to enable it is described [in the VAPIX documentation here](
https://www.axis.com/vapix-library/subjects/t10037719/section/t10036116/display?section=t10036116-t10004224)

It may not be the most easy to use format and it's unfortunately proprietary so not supported by many programs or libraries/frameworks, but it works. I'm aware of a couple of customers who use it for very time critical use-cases so it's accuracy has been tested by at least two parties so it should be reliable.

See the "Timestamp" section for the exact format of the timestamp.

petterwa, 2020-12-09:
Closing this issue since there has not been any activity for 2 weeks.

closed 2020-12-09 by petterwa