AxisCommunications/acap3-examples #45: Where to find allowed TYPES for ax_parameter_add

MikDal002, 2020-11-03:
Hello! 
I am looking for information where I can find allowed types for adding parameters via `gboolean ax_parameter_add ( AXParameter * parameter, const gchar * name, const gchar * initial_value, const gchar * type,
  GError ** error)`. At this moment I found three types: `string`, `int` and `password:string`. I am looking specifically for a type that supports bool which will give me a checkbox in GUI.  I ask because the documentation says *Refer to the SDK documentation for information about parameter types* but I have no idea which SDK.

petterwa, 2020-11-04:
To create a checkbox you can use:  "checkbox:NO,YES"

The allowed parameter types is missing in the current version of the SDK documentation.
From an older version of the SDK:

Radio buttons:
type = "radio:TCPIP|Generic TCP/IP, HTTP|Generic HTTP, OFF|OFF"
Drop-down list (SELECT tag):
type = "enum:TCPIP|Generic TCP/IP, HTTP|Generic HTTP, OFF|OFF"
type = "enum:38400, 57600, 115200, 230400|230400 (experimental)"
Checkbox (or drop-down list depending on configuration). The first value is the unchecked value. If the first value is YES, ON, TRUE, ENABLE* or 1, then reverse order is assumed:
type = "bool:no,yes"
Short forms:
type="yes_no" can be used instead of type="bool:yes,no". type="on_off" can be used instead of type="bool:on,off".
Checkbox (only for boolean values). The first value is the unchecked value. If the first value is YES, ON, TRUE, ENABLE* or 1, then reverse order is assumed:
type = "checkbox:NO,YES"
Plain input field (for string values):
type = "string[:maxlen=X]"
Plain input field (for int values). After the input field, [min..max], [..max], [min..] or [0, 1024..65535] will be displayed. If maxlen is not specified, it will be calculated based on the longest number in the allowed range. The size of the input field will be maxlen + 2 or more in steps of 5:
type = "int[:[maxlen=X;][min=Y;][max=Z]]"
type = "int[:[maxlen=X;]range=0,1024-65535]"
Text area:
type = "textarea"
Plain input field with maximum length 17. Should be used for MAC addresses, for example 00:40:8C:01:34:67.
type = "MACADDR"
Plain input field with maximum length 15. Should be used for IP addresses (numeric value only), for example 123.167.101.145.
type = "IP" or type = "ip"
Plain input field with maximum length 64. Should be used for IP addresses (numeric value) or host names:
type = "IPNAME"
Password input field. The typed characters will be masked by '*' characters:
type = "password[:maxlen=X]"

closed 2020-11-04 by petterwa

MikDal002, 2020-11-09:
Thanks for the comprehensive response. Will it be put into official documentation?

petterwa, 2020-11-10:
Yes, we will add these types to the official documentation for AxParameter api.