AxisCommunications/acap3-examples #47: LAROD_ERROR_LOAD_MODEL

keesschollaart81, 2020-11-06:
I've successfully converted the `SSD MobileNet V2 FPNLite 320x320` model from [here](https://github.com/tensorflow/models/blob/master/research/object_detection/g3doc/tf2_detection_zoo.md) to a quantized TF Lite. Then I converted the TF Lite to Edge TPU:
```
Edge TPU Compiler version 15.0.340273435

Model compiled successfully in 998 ms.

Input model: model.tflite
Input size: 3.72MiB
Output model: model_edgetpu.tflite
Output size: 4.16MiB
On-chip memory used for caching model parameters: 3.42MiB
On-chip memory remaining for caching model parameters: 4.31MiB
Off-chip memory used for streaming uncached model parameters: 0.00B
Number of Edge TPU subgraphs: 1
Total number of operations: 167
Operation log: model_edgetpu.log

Model successfully compiled but not all operations are supported by the Edge TPU. A percentage of the model will instead run on the CPU, which is slower. If possible, consider updating your model to use only operations supported by the Edge TPU. For details, visit g.co/coral/model-reqs.
Number of operations that will run on Edge TPU: 118
Number of operations that will run on CPU: 49
See the operation log file for individual operation details.
```

Now when I run `larodLoadModel(...)` if gives an error with code -2 (LAROD_ERROR_LOAD_MODEL: 'General error for loading model'). Why is this?

It does work when I follow the same conversion path for the `SSD MobileNet v2 320x320` model. Is it the FPNLite? Can I only deploy models that fully run on the TPU, and if so, why?

joha-axis, 2020-11-06:
Hi,

If the difference between the models is the FPN component, it is as you say, likely that the FPN is what causes the issue you're seeing. The Edge TPU supports [a set of operations](https://coral.ai/docs/edgetpu/models-intro/#supported-operations), and while non-TPU supported operations should be handled by mapping these to the CPU rather than crashing, it is possible that it is the cause of the error. You can deploy models with both TPU and CPU operations, but if possible, this should be avoided as the TPU is much faster for most operations.

If you use the `-s` flag when compiling the model using the Edge TPU compiler, you get a list of what operations were mapped to what device, which could be useful to debug the issue. Could you do this for both the working and the non-working model and paste the output here?

Thanks!

keesschollaart81, 2020-11-06:
`ssd_mobilenet_v2_320x320_coco17_tpu-8`:

```
Edge TPU Compiler version 15.0.340273435

Model compiled successfully in 1342 ms.

Input model: model.tflite
Input size: 6.57MiB
Output model: model_edgetpu.tflite
Output size: 6.70MiB
On-chip memory used for caching model parameters: 6.52MiB
On-chip memory remaining for caching model parameters: 1.20MiB
Off-chip memory used for streaming uncached model parameters: 0.00B
Number of Edge TPU subgraphs: 1
Total number of operations: 113
Operation log: model_edgetpu.log

Model successfully compiled but not all operations are supported by the Edge TPU. A percentage of the model will instead run on the CPU, which is slower. If possible, consider updating your model to use only operations supported by the Edge TPU. For details, visit g.co/coral/model-reqs.
Number of operations that will run on Edge TPU: 110
Number of operations that will run on CPU: 3

Operator                       Count      Status

CONCATENATION                  2          Mapped to Edge TPU
DEPTHWISE_CONV_2D              17         Mapped to Edge TPU
DEQUANTIZE                     2          Operation is working on an unsupported data type
CONV_2D                        55         Mapped to Edge TPU
QUANTIZE                       12         Mapped to Edge TPU
LOGISTIC                       1          Mapped to Edge TPU
CUSTOM                         1          Operation is working on an unsupported data type
ADD                            10         Mapped to Edge TPU
RESHAPE                        13         Mapped to Edge TPU
```

`ssd_mobilenet_v2_fpnlite_320x320_coco17_tpu-8`:

```
Edge TPU Compiler version 15.0.340273435

Model compiled successfully in 1050 ms.

Input model: model.tflite
Input size: 3.72MiB
Output model: model_edgetpu.tflite
Output size: 4.16MiB
On-chip memory used for caching model parameters: 3.42MiB
On-chip memory remaining for caching model parameters: 4.31MiB
Off-chip memory used for streaming uncached model parameters: 0.00B
Number of Edge TPU subgraphs: 1
Total number of operations: 167
Operation log: model_edgetpu.log

Model successfully compiled but not all operations are supported by the Edge TPU. A percentage of the model will instead run on the CPU, which is slower. If possible, consider updating your model to use only operations supported by the Edge TPU. For details, visit g.co/coral/model-reqs.
Number of operations that will run on Edge TPU: 118
Number of operations that will run on CPU: 49

Operator                       Count      Status

LOGISTIC                       1          Operation is otherwise supported, but not mapped due to some unspecified limitation
DEPTHWISE_CONV_2D              37         Mapped to Edge TPU
DEPTHWISE_CONV_2D              14         More than one subgraph is not supported
QUANTIZE                       7          Mapped to Edge TPU
QUANTIZE                       3          More than one subgraph is not supported
CONV_2D                        58         Mapped to Edge TPU
CONV_2D                        14         More than one subgraph is not supported
DEQUANTIZE                     1          Operation is otherwise supported, but not mapped due to some unspecified limitation
DEQUANTIZE                     1          Operation is working on an unsupported data type
CUSTOM                         1          Operation is working on an unsupported data type
ADD                            2          More than one subgraph is not supported
ADD                            10         Mapped to Edge TPU
CONCATENATION                  1          More than one subgraph is not supported
CONCATENATION                  1          Operation is otherwise supported, but not mapped due to some unspecified limitation
PACK                           4          Tensor has unsupported rank (up to 3 innermost dimensions mapped)
RESHAPE                        2          Operation is otherwise supported, but not mapped due to some unspecified limitation
RESHAPE                        6          Mapped to Edge TPU
RESHAPE                        4          More than one subgraph is not supported
```
The new 15.x edgetpu_compiler has an expirimental argument for subgraphs:
```
-a, --enable_multiple_subgraphs
       Enable multiple(all) subgraphs (experimental flag).
```
Which I've seen working in other models but throws an error for the fpnlite model which I'll submit in their repo.

joha-axis, 2020-11-09:
Could you try to ssh to the camera and run `journalctl -u larod` and then run e.g., the FPN model? This could potentially give a greater level of detail as to what failed during the loading of the model.

keesschollaart81, 2020-11-09:
Interesting:
```
Nov 09 10:40:33 axis-b8a44f03c3dd larod[1664]: Created a new session ID: 16, client: :1.430, chip ID: 0
Nov 09 10:40:33 axis-b8a44f03c3dd larod[1664]: Session 16: Set chip to 1
Nov 09 10:40:33 axis-b8a44f03c3dd larod[1664]: Session 16: Queuing loading of model LarodRuntime...
Nov 09 10:40:33 axis-b8a44f03c3dd larod[1664]: Session 16: Queuing loading of model LarodRuntime done
Nov 09 10:40:33 axis-b8a44f03c3dd larod[1664]: Input array not provided for operation 'reshape'.
Nov 09 10:40:33 axis-b8a44f03c3dd larod[1664]: Session 16: Could not build an interpreter of the model
Nov 09 10:40:34 axis-b8a44f03c3dd larod[1664]: Session 16: Could not process incoming asynchronous requests: Connection reset by peer
Nov 09 10:40:34 axis-b8a44f03c3dd larod[1664]: Session 16 killed since client's (:1.430) connection has been lost
```

joha-axis, 2020-11-09:
It seems the issue is due to tflite not being able to load the model. The `Input array not provided for operation 'reshape'` message is output by tflite so presumably, getting the model to work involves resolving that error. Thus, it is not directly related to larod. 

Feel free to open a new issue if you run into other problems. Best of luck in your debugging!

closed 2020-11-09 by joha-axis

keesschollaart81, 2020-11-09:
Thanks Joha! True. It would be great though if this underlying error message could be included in the Larod error, in a later version of Larod.

joha-axis, 2020-11-09:
I agree, and thanks to your feedback, it will.

keesschollaart81, 2020-11-13:
@joha-axis, I'm still trying to see how I can get this model to work, you're saying the TFLite interpreter cannot load my model, however in my tests, it does.

This is what I did: 
 
- Download [ssd_mobilenet_v2_fpnlite_320x320_coco17_tpu-8](http://download.tensorflow.org/models/object_detection/tf2/20200711/ssd_mobilenet_v2_fpnlite_320x320_coco17_tpu-8.tar.gz) from the [zoo](https://github.com/tensorflow/models/blob/master/research/object_detection/g3doc/tf2_detection_zoo.md)

- Export a tflite graph (conform [this article](https://github.com/tensorflow/models/blob/master/research/object_detection/g3doc/running_on_mobile_tf2.md))

``` 
export_tflite_graph_tf2.py --ssd_max_detections 100 --pipeline_config_path ssd_mobilenet_v2_320x320_coco17_tpu-8/pipeline.config --trained_checkpoint_dir ssd_mobilenet_v2_320x320_coco17_tpu-8/checkpoint --output_directory ssd_mobilenet_v2_320x320_coco17_tpu-8/tflite_graph
```

- Convert to TFLite:

```py
import tensorflow as tf
import numpy as np

converter = tf.lite.TFLiteConverter.from_saved_model('ssd_mobilenet_v2_320x320_coco17_tpu-8/tflite_graph/saved_model')

def representative_dataset_gen():  # rewrite this for real data.
    for _ in range(500):
        yield [np.random.uniform(0.0, 1.0, size=(1, 320, 320, 3)).astype(np.float32)]
    
converter.experimental_new_converter = True
converter.representative_dataset = representative_dataset_gen
converter.optimizations = [tf.lite.Optimize.DEFAULT] 
converter.target_spec.supported_ops = [tf.lite.OpsSet.TFLITE_BUILTINS_INT8, tf.lite.OpsSet.SELECT_TF_OPS] 
converter.inference_input_type = tf.uint8
converter.allow_custom_ops = True
tflite_model = converter.convert()

with open('ssd_mobilenet_v2_320x320_coco17_tpu-8/model.tflite', 'wb') as f:
    f.write(tflite_model)
```

- Running the [the Python based interpreter](https://www.tensorflow.org/lite/guide/inference#load_and_run_a_model_in_python) with this code:

```py
import numpy as np
import tensorflow as tf

interpreter = tf.lite.Interpreter(model_path="ssd_mobilenet_v2_fpnlite_320x320_coco17_tpu-8/model.tflite")
interpreter.allocate_tensors()

input_details = interpreter.get_input_details()
output_details = interpreter.get_output_details()

input_shape = input_details[0]['shape']
input_data = np.array(np.random.random_sample(input_shape), dtype=np.uint8)
interpreter.set_tensor(input_details[0]['index'], input_data)

interpreter.invoke()

output_data = interpreter.get_tensor(output_details[0]['index'])
print(output_data)
```
And the interpreter is able to run the model. What did you do different to get to your result? Is it the embedded context?

joha-axis, 2020-11-16:
Judging from the diversity of status messages you get when performing the edge TPU compilation of the model, I would assume that's your issue, which then translates to a model which cannot be loaded. 

For debugging purposes, you can try running the model on the CPU instead of the TPU, i.e. packaging the non-edge TPU compiled tflite-model in the ACAP and changing the chip that larod uses to 2 (= CPU, 4 = TPU) within your application. If that works, the problem lies in the edge TPU pipeline.