AxisCommunications/acap3-examples #48: AxHttp access level

MikDal002, 2020-11-10:
Hello! 

Thanks again for the fast response to my previous issue. Now I have a question about the access level in AXHttp library. I can see there three levels: **viewer**, **operator** and **administrator** unfortunately in the documentation I can't see any differences and when trying to make an HTTP request all levels require providing login and password for root user. Is there any value that would allow me to send requests without any authentication within the camera? I need that to integrate two ACAP applications but one of them, which is the source of events, is a 3rd party app and doesn't have support for authentication.

petterwa, 2020-11-11:
Hi,
we don't have any github example yet for AxHttp, so please check the ACAP developer pages for more info on this api https://www.axis.com/developer-community. Check VAPIX documention for more info about access levels and corresponding user management in (pwdgrp.cgi) to avoid using root user, but I don't think there is any option in AxHttp to completely avoid authentication.

MikDal002, 2020-11-19:
Thanks for the response, I resolved my problem with the TCP server. Could you tell me where I can suggest an improvement to the AXIS SDK? For example, I would like to suggest extending the parameters library to give developers the possibility to make groups of app's settings (the same way as in the settings of the camera). I am working on an app that takes many properties and I don't want to spend my time doing a special configuration page.

petterwa, 2020-11-20:
Hi,
I'm happy to hear that the problem with TCP server was gone. 
We are very interested in suggestions and ideas to improve the axis sdk. You can use this github issue mechanism also for ideas and suggestions. It's good if you report suggestions  as separate issues.

petterwa, 2020-12-15:
Closing this issue since the original issue is solved and the suggestion has moved to a dedicated issue.

closed 2020-12-15 by petterwa