AxisCommunications/acap3-examples #55: Device unique ID

MikDal002, 2020-12-11:
Hello, I have been looking for a long while how to get the device's id on Linux, but at this moment I am stuck because I haven't found any common way. But I noticed that a serial number is shown on the camera's page (in the Information window). Is there any way to get that with C API?

petterwa, 2020-12-15:
Serial number is currently only exposed via Vapix network api, see "SerialNumber" in basicdeviceinfo.cgi and has unrestricted access. There is a gap in ACAP api's when it comes to the serial number and more people has been asking for it.
I'm not aware of any work around except using Vapix to retrieve serial number.

pataxis, 2023-10-11:
We have recently launched a new way in ACAP Native SDK (part of ACAP 4 offering) to make [VAPIX requests from an ACAP application](https://axiscommunications.github.io/acap-documentation/docs/develop/VAPIX-access-for-ACAP-applications.html), e.g. to basicdevicenfo.cgi that was mentioned above. Migrating to ACAP Native SDK would therefore be a way to achieve this. See example applications in [acap-native-sdk-examples](https://github.com/AxisCommunications/acap-native-sdk-examples).

closed 2023-10-11 by pataxis