AxisCommunications/acap3-examples #56: Expand AxParameters library

MikDal002, 2020-12-11:
Hi, as I mentioned in #48 I would like to suggest a small improvement: 

> I would like to suggest extending the parameters library to give developers the possibility to make groups of app's settings (the same way as in the settings of the camera). I am working on an app that takes many properties and I don't want to spend my time doing a special configuration page.

Actually, the AxParameters library has all types I need, the only problem is that the list of my app's properties is quite long (and will grow in the future). So, to make all properties easier to discover and manage I would like to get the possibility to group them. 

What do you think about that?

petterwa, 2020-12-15:
Thank you for submitting this suggestion. 

It makes sense to be able group application parameters, but I'm not sure if we should extend AxParameter API since it rely on param.cgi and we might seek other solutions/enablers for applications with many parameters.

pataxis, 2023-10-11:
The suggestion was forwarded to the team and has been noted for future implementations of parameter APIs.

closed 2023-10-11 by pataxis