AxisCommunications/acap3-examples #57: Could not allocate tensors

nejcmedved, 2020-12-15:
I am getting an issue with running the example

```
/tensorflow_to_larod /usr/local/packages/tensorflow_to_larod/
model/converted_model.larod 128 128 1 -c 2
```

`setupLarod: Unable to load model: Could not load model: Could not allocate tensors`

if I run it on chip 4 as in example it says

```
2025-05-22T11:46:03.281+02:00 axis-b8a44f087b50 [ INFO    ] tensorflow_to_larod[2054]: Starting ...
2025-05-22T11:46:03.291+02:00 axis-b8a44f087b50 [ INFO    ] tensorflow_to_larod[2054]: chooseStreamResolution: We select stream w/h=240 x 160 based on VDO channel info.
2025-05-22T11:46:03.292+02:00 axis-b8a44f087b50 [ INFO    ] tensorflow_to_larod[2054]: Creating VDO image provider and creating stream 240 x 160
2025-05-22T11:46:03.292+02:00 axis-b8a44f087b50 [ INFO    ] tensorflow_to_larod[2054]: Dump of vdo stream settings map =====
2025-05-22T11:46:03.313+02:00 axis-b8a44f087b50 [ INFO    ] tensorflow_to_larod[2054]: Setting up larod connection with chip 4 and model /usr/local/packages/tensorflow_to_larod/model/converted_model.larod
2025-05-22T11:46:03.326+02:00 axis-b8a44f087b50 [ ERR     ] tensorflow_to_larod[2054]: setupLarod: Could not select chip 4: Could not find chip 4
```

joha-axis, 2020-12-15:
Hi,

To use chip 4 (the Edge TPU), you need an Edge TPU equipped camera, such as the Q1615 Mk3. What camera are you running the application on? 

Regarding the tensor allocation error, this is likely due to that the application is trying to use an Edge TPU-compiled model with the CPU (chip 2). Thus, if you want to run the model on the CPU, you may resolve this issue by using a model that has not been compiled to run on the Edge TPU, which corresponds to not performing step 4 in the [Quickstart](https://github.com/AxisCommunications/acap3-examples/tree/master/tensorflow-to-larod#quickstart) section of the example.

nejcmedved, 2020-12-15:
Hi thanks a lot,

I assume you work for Axis. First thanks a lot for all the examples! I am still getting familiar with Docker though. I work for one Slovenian company in traffic sector and I am looking forward to write software for Axis cameras.

I couldn't find which cameras has Edge TPU chips equipped.

Will skip step you mentioned and try again :)

Best, Nejc

nejcmedved, 2020-12-16:
Hi again,

I have tried now with converted_model.tflite and the error is below:

```
2025-05-23T06:53:57.703+02:00 axis-b8a44f087b50 [ INFO    ] tensorflow_to_larod[9271]: Setting up larod connection with chip 2 and model /usr/local/packages/tensorflow_to_larod/converted_model.tflite
2025-05-23T06:53:57.735+02:00 axis-b8a44f087b50 [ ERR     ] tensorflow_to_larod[9271]: setupLarod: Unable to load model: Could not load model: basic_string::_M_construct null not valid
```

do you have any idea about this behavior?

Best, Nejc

joha-axis, 2020-12-16:
I'm glad you like the examples!

It is hard to tell what the cause of the issue is from the error alone, but as the problem arises when loading the model, either your model or the input to the setupLarod method is likely the culprit. In order to find out what the issue might be, you could try to look at how these items differ from their counterparts in the example.

I'm going to close this issue now as we've moved passed its topic. Feel free to open a new issue if needed!

closed 2020-12-16 by joha-axis