AxisCommunications/acap3-examples #60: Get Tensor Details

keesschollaart81, 2021-01-11:
The Tensorflow Lite interpreter (Python) has some operations to get details of input/output tensors. Im my case, the models tensors contain metadata in the `quantization` and `quantization_parameters` properties of the Tensor that I need. I cannot find a Larod operation that allows me to retrieve these values.

In the TFLite Python Interpreter, the tensor metadata can be obtained using `interpreter.get_input_details()` and `interpreter.get_output_details()` documented [here](https://www.tensorflow.org/api_docs/python/tf/lite/Interpreter#get_input_details).

I guess [this SO answer](https://stackoverflow.com/a/57213376) might be helpful here as well.

joha-axis, 2021-01-11:
Hey,

All tflite quantization properties are not available to get through larod right now, but this is likely to change. In the meantime, the easiest workaround is to retrieve them through tflite before converting the model to .larod. Will this solution work for you?

keesschollaart81, 2021-01-11:
Yes, that is what I do now but good to know this improvement is on your radar.

closed 2021-01-11 by joha-axis