AxisCommunications/acap3-examples #61: Error setting AxParameter value

TiboCochl, 2021-01-15:
Hello,

I've been unable to set the value of my AxParameters for my application under the camera webpage. After some research I found that the problem was that my appname contains an underscore (eg : foo_bar). Therefore when the request axis-cgi/param.cgi?action=update is done I receive the following message :
# Error: Error setting 'root.Foo.bar.AudioSource0' to 'test'
As you can see the underscore is replaced by a dot misleading the process saving the data to the param.conf

I think it would be nice to restrict the app name to not use underscore or even better fix directly that underscore transformation.

petterwa, 2021-01-21:
Hi,

I checked with the experts on AxParam API and this is caused by a limitation in an old component so I don’t think we will succeed to get rid of the root cause. 

I think we need to go with the workaround and restrict application name to not use underscore as you suggest.

TiboCochl, 2021-01-22:
Thank you for the update ! Closing this now

closed 2021-01-22 by TiboCochl