AxisCommunications/acap3-examples #62: Creation of new TPU model

AriTelias, 2021-01-20:
Hello Axis,
 
We are working on an ACAP application for the Q1615 Mark 3, using AI.
We are following the instructions found in the official Axis repo.
https://github.com/AxisCommunications/acap3-examples/tree/master/vdo-larod

We are able to use the simple TPU detection model suggested there (detect.larod at 300x300), however we have not been able to compile our own TPU models with higher resolutions or different aspect ratios.
Is there any recommendation or guidance from the AXIS AI team?

Which AI architecture should we train our model on? (MobileNet V2, MobileDet, YOLO, etc...)
Which compiler and quantization technique?

Thanks and regards,

--------------------------------------------------------

The issue with TPU model creation is: "edgetpu_compiler fails for higher resolution or 16:9 aspect ratio fully quantized model with
error message - Internal compiler error. Aborting!"
Edge TPU Compiler version: 15.0.340273435
Tensorflow version: 1.15.0
Resolutions of the model: 480 x480, 480x270, 400x225, 340x340
Architecture: SSD_MOBILEDET_EDGETPU - Int8 Quantize Aware Training
Tflite converter used: TOCO

joha-axis, 2021-01-28:
Hi,

There is an example called [tensorflow-to-larod](https://github.com/AxisCommunications/acap3-examples/tree/master/tensorflow-to-larod) which goes through the process of creating your own simple model in Tensorflow and converting it to be used on a camera with the Edge TPU. You can use this as a guide to the conversion process as it's the same no matter if you are using a simple model like the one in the example or Tensorflow object detection model.

Using this example, I created model with an 480x270x3 input and compiled it for the Edge TPU and ran it without seeing any issues. A difference in the example's environment versus yours is that you are using Tensorflow 1.15. Later versions of Tensorflow have improvements for tflite conversion and quantization. Thus, if you could test your setup with e.g., Tensorflow v2.3.0 (or directly in that example's environment), that would be great.

Regarding what architecture to use, I'm afraid I cannot give an answer as it depends on your use-case. However, you can find pretrained models that are already compiled for the Edge TPU on the [Coral site](https://coral.ai/models/), which allows you to try out many different architectures quite easily.

Regards,
Johan

AriTelias, 2021-01-28:
Hi Johan,
Hope all is well in Sweden!
Thank you for your reply - we are indeed following the examples for Larod conversion, but it could indeed be that the Tensorflow version is affecting our  model creation.
We will follow your recommended steps and update here if we have relevant findings.
have a great day,
The AiDANT Team

joha-axis, 2021-03-01:
I am closing this issue for now as there has been no activity for the past month. Feel free to open a new issue if you run into further problems.

closed 2021-03-01 by joha-axis