AxisCommunications/acap3-examples #63: How to get rtsp link inside of Axis Encoder?

huysc0ville, 2021-01-27:
I'm trying to push the stream to AWS Kinesis by Gstreamer inside of the Axis Encoder. But one problem is I don't know what exactly the RTSP link from inside of the Axis Encoder? Could you tell me how could I get the RTSP link?

johan-bjareholt, 2021-01-28:
If I understand correctly, you are trying to write an ACAP which should have video over RTSP as the source and wants to forward it to AWS Kinesis?

To connect over RTSP inside an ACAP should be the same URL as when connecting to it from the outside, you just need to change the ip to either 127.0.0.1 or to localhost.
How to connect over RTSP and the URL is documented in the VAPIX documentation here:
https://www.axis.com/vapix-library/subjects/t10037719/section/t10051110/display?section=t10051110-t10003610

So for example, you could try `rtsp://127.0.0.1/axis-media/media.amp?video=1`

Does that answer your question?

huysc0ville, 2021-02-02:
@johan-bjareholt  That's exactly what I want. Thank you very much.

closed 2021-03-02 by petterwa