AxisCommunications/acap3-examples #67: How to include/import or link third-party libraries into one ACAP application?

huysc0ville, 2021-02-18:
Now, I'm working on a project, using Axis Encoder to send video stream to AWS Kinesis Video Stream (KVS). So I want to implement an ACAP Application to handle it.  The KVS service has a Kinesis Producer SDK (written by C). I build the SDK to a library by CMake. 
Now I want to include/link and import that library files into our ACAP application and use them in my source code (also in C). But I don't know what is exactly the steps and how I need to do to get it to work. 
Could you tell me how I could include/import one third-party library into an ACAP application?

petterwa, 2021-02-19:
We have added a new example that shows how to include opencv libraries into the ACAP application and I think same pattern can be used when including other third party libs.
https://github.com/AxisCommunications/acap3-examples/blob/master/using-opencv/app/Makefile

huysc0ville, 2021-02-20:
@petterwa thank you very much. Btw, I already contacted the support team on the Help desk page, and they sent me this sample source code. It doesn't have any sensitive information, so I attach it here, hope it could help other ones also. I also close this ticket here. 
[local_shared_lib.zip](https://github.com/AxisCommunications/acap3-examples/files/6015114/local_shared_lib.zip)

theowlfla, 2021-02-20:
Thanks a mill that's brilliant. Really appreciate it.

closed 2021-04-14 by mikaelli-axis