AxisCommunications/acap3-examples #69: AXHttp - Querystring for non-GET requests

keesschollaart81, 2021-03-08:
I'm using the AXHttp, lets say I have a product.cgi which I would like to use in a RESTfull way with routes like:

[POST] /local/MyApp/product.cgi
[GET] /local/MyApp/product.cgi?id=1
[PUT] /local/MyApp/product.cgi?id=1

Is it correct that the last PUT call cannot be implemented because it mixes QueryString parameters with a body payload? And if not, how do I access querystring parameters when a request payload is send?

mikaelli-axis, 2021-04-01:
Just wanted to acknowledge that we have seen this question and are working on an answer. Additionally, we are aware of the limitations of the axHttp API and are looking at better ways to support HTTP endpoints in the future.

petterwa, 2021-04-07:
The AxHttp interface should not be seen as a standard HTTP endpoint, but rather a simple and limited CGI plugin mechanism that is tailor made for a few specific use cases.
When it comes to HTTP request methods, AxHttp implements GET and POST, but not HEAD, PUT or DELETE. Hence PUT should not be used at all in combination with AxHttp. 
Http headers are parsed in a internal layer below AxHttp interface, but they are unfortunately not exposed in the API.

I hope we can came back soon with a better solution for severing HTTP requests.

keesschollaart81, 2021-04-08:
> but rather a simple and limited CGI plugin mechanism that is tailor made for a few specific use cases

Clear. I'll deal with that, for now

> I hope we can came back soon with a better solution for severing HTTP requests. 

Would be awesome, please report back here by then

closed 2021-04-08 by keesschollaart81