AxisCommunications/acap3-examples #70: Accessing camera’s built-in motion detection events in ACAP

qraleq, 2021-03-14:
Hi, 

I'm interested in whether it is possible to get the output of the camera's motion detection events using ACAP in real-time as they occur and use them in a custom application? 

I saw the OpenCV example in this repo that performs motion detection using BGS (https://github.com/AxisCommunications/acap3-examples/tree/master/using-opencv), and that is really great, but I'm interested if it is possible to leverage camera’s built-in motion detection to reduce the processing requirements.

Thank you for your help!

petterwa, 2021-03-26:
Sorry for slow response

There is not yet any public ACAP API for accessing the built in motion object detection events. There are plans to publish such ACAP API but I'm not sure when it will be available. 

Motion detection event are otherwise available from the network api in the analytics RTSP stream, but I think it's quite messy to use it from an ACAP context.

qraleq, 2021-03-26:
Hi, @petterwa! Thank you very much for your answer. We're looking forward to the moment when this API will be exposed since it will enable cool features to be built on top of the basic motion detector. 

As for the analytics RTSP stream, how do you access this? Can you please share some instructions?

Thank you!

petterwa, 2021-03-26:
Hi,

According to the team working with this you should be able to get the RTSP stream with motion detection events with this oneliner gst-launch-1.0 rtspsrc location='rtsp://root:<PASSWORD>@<CAMERA_IP>/axis-media/media.amp?video=0&audio=0&event=off&analytics=polygon' ! fdsink

Note. If you have very old firmware on your camera there might be needed additional steps to unlock the stream.

qraleq, 2021-03-29:
Hi, @petterwa! I tested this, and it works really nice. Thank you!

I was wondering if there is additional documentation on this URL scheme that you're using. Can you get other events such as license plate metadata in the RTSP stream?

petterwa, 2021-04-08:
Hi,
 
Please check "Event data streaming" in the Vapix documentation for more info on the RTSP stream: 
https://www.axis.com/vapix-library/subjects/t10037719/section/t10038219/display?section=t10038219-t10029806

closed 2021-04-21 by mikaelli-axis