AxisCommunications/acap3-examples #72: Larod - Firmware 10.4.x issue for larodLoadModel

keesschollaart81, 2021-03-25:
Yesterday I discovered that our app does not work anymore with the latest 10.4.5 firmware for the P3255 LVE camera (last known firmware to work is 10.3.?). I have not yet upgraded my Q1615 as I don’t want to block myself for further developments.

It seems that larodLoadModel now returns a zero pointer but the larodError is also a zero pointer. In a few random some cases I get a ‘Segmentation fault (core dumped)’. 

This is all code that runs perfectly fine on previous firmware’s… Did I miss a release note?

keesschollaart81, 2021-03-25:
I notice on the 10.3 firmware, there is a `/usr/lib/liblarod.so.1.14.7` and on the 10.4 there is a `/usr/lib/liblarod.so.2.1.26`...

petterwa, 2021-03-25:
The Larod library in the 10.4.5 firmware should work with applications built with SDK 3.2 even though the the library version number has been stepped up. I will notify the team working with larod about this problem.

keesschollaart81, 2021-03-25:
Thanks! Let me know if there's anything I can do to provide you with more data to debug this.

keesschollaart81, 2021-04-08:
Thanks to a mail conversation with an engineer of Axis we found the root cause. The 10.4.5 release contains the 2.x release of Larod. It is compiled in a way that is backwards compatible for ACAP app developers: when compiling C code. However, I'm using C# and I invoke these api's runtime using native bindings/interop/PInvoke. It turns out, the `/usr/lib/liblarod.so.2.1.26` has implementations for both the new 2.x api's as well for the 1.x api's. .NET does not know how to distinct these 'semantic versioned' signatures and takes the latest. As my code expected the 1.x signatures, if crashes with a runtime exception. Now that I have the header file for the 2.x version of Larod, I ported the code of my app and now it works again.

closed 2021-04-08 by keesschollaart81

petterwa, 2021-04-09:
Hi,

C# was a bit of a surprise and we didn't consider this in the 1.x backward compatibility solution for larod.
There is a beta release of the ACAP3 SDK on dockerhub that contains version 2.x of larod header files and libs, see axisecp/acap-sdk:3.3_beta1-armv7hf-ubuntu20.04 
https://hub.docker.com/repository/docker/axisecp/acap-sdk

keesschollaart81, 2021-04-09:
Thanks! I know there are not much apps (yet) written in languages other than C. That's why I did as much as I could to investigate myself. As I don't depend on the SDK to build my app, I'm not using the Docker container on a daily basis. It is a good reminder though that it is the place where I can find the latest header files! 

Maybe, one day, we can run ACAP apps in a docker container with Axis-build language bindings.... That would be 🤩