AxisCommunications/acap3-examples #73: Possible for app to access RTSP stream without a username/password account?

mms-, 2021-03-27:
I'd like my app to make use of the RTSP stream, however it seems to requires the user to input the username and password. Is it possible for the app to access the rtsp stream without username and password? 

i.e. local rtsp server detects that the connection is coming from local host and allows the connection without password requirement.

mikaelli-axis, 2021-04-01:
It is currently not possible to bypass the HTTP authentication in the way you describe above without manually editing the Apache configuration in the camera, which I would only recommend for testing purposes. However, accessing VAPIX from an ACAP in a secure manner is something we are looking into supporting, though I cannot share anything about the timeline at this point.

mms-, 2021-04-01:
I understand. Currently we are mostly in need of RTSP port 554 access however VAPIX would of course also be useful.

As is, we are forced to require end users to input their camera username and password into our app! so our app can access the rtsp stream (APIs can't be used for our purpose.) This of course is bad however the only other choice is to run our own RTSP server within the app which is equally silly.

The rtsp/vapix servers would need to infer the calling app identity somehow and give access based on the app permissions (all apps have equal rights currently?) Maybe can be done by the server linking an incoming request to an on camera app process or the app includes a token with each request. Though tokens could leak then external requests could make calls under the app identity, so process based approach may be preferred.

mikaelli-axis, 2021-04-06:
That is correct, currently all ACAPs run as the sdk-user. However, as of AxisOS 10.5 / ACAP SDK 3.3 we are introducing application specific users so that files, processes and permissions can be application specific. We hope to soon release official documentation about this and if you are part of the Axis ADP program you might already have had some information.

closed 2021-04-21 by mikaelli-axis

mikaelli-axis, 2021-04-23:
Yes there are a number of camera models that due to hardware limitations cannot run newer version of AxisOS. Discussion on firmware upgrade is beyond our ACAP examples, so I would direct further questions about this topic to our official support channels. 

However, if you are proficient with Apache, you could potentially solve this by adding to the web server configuration through the ACAP post installation script. Normally this is something we recommend against doing, as firmware upgrades can cause issues with such modifications, but may provide a way forward for devices with limited firmware upgrades.

mms-, 2022-12-14:
Hi, any solution to allow apps to access the local VAPIX API and RTSP server streams without need for passwords? i.e. based on permissions?

mikaelli-axis, 2022-12-15:
We are looking into allowing ACAPs to get permissions to access VAPIX APIs and hope to share information about this as soon as we have a more solid time plan.

mms-, 2022-12-15:
Thank you. What about access to the rtsp stream? Duplicating this in our app not ideal for CPU resources. Our app has its own authentication and exposes rtsp for consumption by other parts of the system.

mikaelli-axis, 2022-12-16:
Not sure I understand exactly what you are asking. Can you elaborate on what you would like to do / what you want to avoid?

mms-, 2022-12-16:
Our app consumes the existing on camera rtsp stream. There are reasons why this representation works better for us compared to the more raw stream through the ACAP media capture API's.

Currently on app installation, users need to provide a username and password into our app so it can make a request to rtsp://localhost..  This makes installation much harder and brittle long term as passwords change.

Ideally any app would have automatic access to the rtsp stream because it already has access to same image data (though different format) through ACAP.  So the credentials in this case are redundant.

And if automatic rtsp stream access is a security issue then why not have permissions per app, this way the user just flicks a checkbox on app install or assigns a pre existing role or user that has access. Could be like this for VAPIX though rtsp seems redundant. VAPIX because the app still needs to use these API's as not everything is exposed through ACAP and asking users for passwords is very cumbersome and outdated.

Thanks!