AxisCommunications/acap3-examples #74: How to discover other AXIS cameras on the network from ACAP?

WilfR, 2021-03-30:
I understand that it is possible in principle to discover AXIS cameras on the local network - this is what AXIS Device Manager does after all. Is it possible to implement this functionality in an ACAP application, and if so, how? That is, can I write an ACAP application that discovers other AXIS cameras on the network and then, for example, writes their IP addresses into the ACAP application's local log file? 
Thanks, 
Wilf.

mikaelli-axis, 2021-04-01:
Axis cameras implement both Bonjour and UPnp (configuration available under TCP/IP in the web UI), but we currently don't offer any clients for these discovery mechanisms in the camera as part of the ACAP platform. If you want to use these methods to find other Axis products on the network, you would have to include a client, such as avahi (https://www.avahi.org/) in your application. As a side note, broadcast discovery methods are not 100% reliable in corporate networks as many IT setups filter them out at the switch level to limit network noise.

WilfR, 2021-04-14:
Thanks very much Mikael. We will follow up with your suggestions.

closed 2021-04-14 by WilfR