AxisCommunications/acap3-examples #75: Saving persistent data in ACAP filesystem

WilfR, 2021-04-08:
I would like to save a small amount of historical data from my ACAP application (a few numbers per day). I know that I could create a file under "/tmp" and append new data to that file but I am worried that "/tmp" gets cleaned out if the camera restarts. However I cannot find a non-temporary location in the camera filesystem that my ACAP has permission to write to. How can I store persistent data into the ACAP filesystem and be (relatively) confident that the data won't be lost if the camera is reset or restarted? 
Thanks very much. 
Wilf

mikaelli-axis, 2021-04-13:
I recommend you take a look at the ACAP local data folder as a way to save persistent data. Check out the documentation in the online manual: https://www.axis.com/products/online-manual/s00004#t10152934 .

WilfR, 2021-04-14:
Thank you very much Mikael - it looks like the localdata folder is exactly what I was looking for. Somehow I missed it (or forgot about it!) in the documentation. My apologies for the wasted motion, but I appreciate your kind assistance.

closed 2021-04-14 by WilfR