AxisCommunications/acap3-examples #77: OpenCV with OpenCL on Artpec-7

nejcmedved, 2021-04-21:
Dear all,

I have an issue with OpenCL with ARTPEC-7 camera. Is there any settings that need to be enabled on camera?

under /dev I see camgpu device. Is that OpenCL device?

Best, Nejc

petterwa, 2021-04-21:
Hi,

OpenCL is enabled by default on artpec 7, but there has been problems in the past with access to all dependencies for OpenCL for the sdk user.  Which camera and firmware are you using?

The openCL device is /dev/galcore and sdk user should have access to this device via gpu group as well as the libs and icd file in 
/etc/OpenCL/vendors.

nejcmedved, 2021-04-21:
hi,

camera is M1135

Current firmware version: 10.2.0

Best, Nejc

nejcmedved, 2021-04-21:
I have upgraded to firmware to 10.4.0 and still no /dev/galcore nor /etc/OpenCL visible not under root nor sdk

petterwa, 2021-04-22:
Hi,

I checked M1134 and I'm also missing driver and libs for OpenCL.

According to Graphics team the GPU is enabled on almost all Artpec 7 products, but there are a few exceptions -  M11134, M1135 and P7304 where GPU is disabled because the memory limitations on these products. 
I.e. there is no support for OpenCL on your M1135 device.

nejcmedved, 2021-04-22:
Hi,

thanks for time invested :) 

Is there any documentation about Axis cameras? Which ones has OpenCL and which ones has Coral AI chips?

what I could find on Axis website is only which cpu's are inside

Best, Nejc

nejcmedved, 2021-04-22:
Would any of those be equipped with either OpenCL or Coral AI?

M1137-E
P1375-E

Best, Nejc

petterwa, 2021-04-22:
Hi, 

both M1137-E and P1375-E are Artpec 7 products with GPU enabled, so OpenCL should be supported. 
They don't have a dedicated DLPU/TPU, but you can use Larod on CPU on these products.  
DLPU/TPU is currently only available in Q1615-MkIII.

We currently just have a general statement in the ACAP documention (https://www.axis.com/products/online-manual/s00004#t10158685) about OpenCL support on Artpec-7, but it's not visible in the product manuals.

closed 2021-04-23 by petterwa