AxisCommunications/acap3-examples #84: make failed for 'object_detection'

fangliang425, 2021-05-14:
Hi, 

When I simply ran the 1st step `./build_acap.sh object_detection_acap:1.0`, got the error, it seems like there are some error in object_detection.c, or did I do something wrong?
[image]

joha-axis, 2021-05-17:
Hey, 

It seems this was caused by a missing version string in the Makefile of the object-detection example, which caused the wrong version of larod to be used during compilation. I've updated the file with a fix for this, so performing the build now should work.

closed 2021-05-17 by joha-axis

fangliang425, 2021-05-17:
``> Hey,
> 
> It seems this was caused by a missing version string in the Makefile of the object-detection example, which caused the wrong version of larod to be used during compilation. I've updated the file with a fix for this, so performing the build now should work.

Thanks, it works. 
However, after I installed the 'object_detection' application and checked the log on browser, the app failed to launch. Still doesn't work after reboot.

```
----- Contents of SYSTEM_LOG for 'object_detection' -----

2021-05-17T14:36:38.680+03:00 axis-b8a44f224c2d [ INFO    ] object_detection[0]: starting object_detection
2021-05-17T14:36:38.726+03:00 axis-b8a44f224c2d [ INFO    ] object_detection[4432]: Starting ...
2021-05-17T14:36:38.860+03:00 axis-b8a44f224c2d [ INFO    ] object_detection[4432]: Creating VDO image provider and creating stream 480 x 300
2021-05-17T14:36:38.860+03:00 axis-b8a44f224c2d [ INFO    ] object_detection[4432]: Dump of vdo stream settings map =====
2021-05-17T14:36:38.860+03:00 axis-b8a44f224c2d [ INFO    ] object_detection[4432]: chooseStreamResolution: We select stream w/h=480 x 300 based on VDO channel info.
2021-05-17T14:36:39.299+03:00 axis-b8a44f224c2d [ INFO    ] object_detection[4432]: Setting up larod connection with chip 4 and model /usr/local/packages/object_detection/model/converted_model.larod
2021-05-17T14:36:39.299+03:00 axis-b8a44f224c2d [ INFO    ] object_detection[761]: Last message 'Dump of vdo stream s' repeated 1 times, suppressed by syslog-ng on axis-b8a44f224c2d.xxx.xxx
2021-05-17T14:36:52.752+03:00 axis-b8a44f224c2d [ ERR     ] object_detection[4432]: setupLarod: Could not connect to larod: Could not create session: Message recipient disconnected from message bus without replying
```

joha-axis, 2021-05-17:
What camera model and camera firmware version are you running?

fangliang425, 2021-05-17:
Hi,
Camera model: AXIS Q1615 Mk III Network Camera
Firmware version: 10.5.2

joha-axis, 2021-05-20:
No, there should not be any steps needed to be done due to how you reboot. 

I suspect that this might be due to your camera's configuration or other applications running simultaneously as I am not able to reproduce it myself on the same setup. Could you reboot the camera, SSH to it and run `journalctl -f` and then start the application? This should output logs that could indicate what is causing this error.

fangliang425, 2021-05-20:
> No, there should not be any additional steps needed to be done due to how you reboot.
> 
> I suspect that this might be due to your camera's configuration or other applications running simultaneously as I am not able to reproduce it myself on the same setup. Could you reboot the camera, SSH to it and run `journalctl -f` and then start the application? This should output logs that could indicate what is causing this error.

Hi, thanks for quick reply. I just rebooted it and ran `journalctl -f`, some logs of failure and error output in loop (did not start app yet). It seems to fail to launch the TPU device.

```
May 20 10:49:01 axis-b8a44f224c2d.visy.fi objectanalytics[1429]: The MQTT client objectanalytics failed to connect, cause TCP/TLS connect failure
May 20 10:49:04 axis-b8a44f224c2d.visy.fi objectanalytics[1429]: The MQTT client objectanalytics failed to connect, cause TCP/TLS connect failure
May 20 10:49:05 axis-b8a44f224c2d.visy.fi viewarea[1171]: main.c:306 [run_service] Terminating...
May 20 10:49:05 axis-b8a44f224c2d.visy.fi systemd[1]: viewarea-dbus.service: Succeeded.
May 20 10:49:06 axis-b8a44f224c2d.visy.fi api-discovery[1280]: No request received, starting termination process
May 20 10:49:06 axis-b8a44f224c2d.visy.fi api-discovery[1280]: DBus name released
May 20 10:49:06 axis-b8a44f224c2d.visy.fi systemd[1]: api-discovery.service: Succeeded.
May 20 10:49:07 axis-b8a44f224c2d.visy.fi objectanalytics[1429]: The MQTT client objectanalytics failed to connect, cause TCP/TLS connect failure
May 20 10:49:11 axis-b8a44f224c2d.visy.fi larod[1539]: Could not create service: Could not initialize TPU
May 20 10:49:11 axis-b8a44f224c2d.visy.fi systemd[1]: Condition check resulted in Crash dump disk detector being skipped.
May 20 10:49:11 axis-b8a44f224c2d.visy.fi systemd[1]: larod.service: Main process exited, code=exited, status=1/FAILURE
May 20 10:49:11 axis-b8a44f224c2d.visy.fi systemd[1]: larod.service: Failed with result 'exit-code'.
May 20 10:49:11 axis-b8a44f224c2d.visy.fi kernel: apex 0000:01:00.0: RAM did not enable within timeout (12000 ms)
May 20 10:49:11 axis-b8a44f224c2d.visy.fi kernel: apex 0000:01:00.0: Error in device open cb: -110
May 20 10:49:11 axis-b8a44f224c2d.visy.fi kernel: apex 0000:01:00.0: Apex performance not throttled due to temperature (-89700 mC)
May 20 10:49:11 axis-b8a44f224c2d.visy.fi objectanalytics[1429]: 273: Could not connect to larod: Could not create session: Message recipient disconnected from message bus without replying
May 20 10:49:11 axis-b8a44f224c2d.visy.fi objectanalytics[1429]: Larod setup failed

```

joha-axis, 2021-05-20:
Hi,

Thanks for the logs. This issue requires the assistance of the AXIS Helpdesk. Reach out to them through this link: https://www.axis.com/support/helpdesk

fangliang425, 2021-05-20:
> Hi,
> 
> Thanks for the logs. This issue requires the assistance of the AXIS Helpdesk. Reach out to them through this link: https://www.axis.com/support/helpdesk

Thanks, I'll seek help there.

commit 79dba748 (2021-05-17): fix incorrect larod version usage (#85)

Fixes #84