AxisCommunications/acap3-examples #87: Object detection

juanfeyero, 2021-05-19:
Hello,
I have followed the instruction to generate the .eap file and installed it on a AXIS Q1615 Mk III camera. However, when I start the app, it never change the status to "running" and the logs don't provide any detection of the object. I tested the TensorFlow to larod app and that works.

Could you provide a .eap to test it or tell me what happens?

joha-axis, 2021-05-20:
Hey,

Can you provide us with the firmware version you are running as well as the journalctl logs? You can get these logs by SSH:ing to the camera, running `journalctl -f` and then starting your application. This should give us a better idea of what is causing the issues.

juanfeyero, 2021-05-21:
**AXIS Q1615 Mk III Network Camera**
**Firmware** 10.0.2
I could not find how to activate the SSH, instead I used serverreport.cgi
**Logs:**
2021-05-20T22:40:04.495-06:00 axis-b8a44f13e385 [ INFO    ] systemd[1]: fwmgr.service: Succeeded.
2021-05-20T22:40:26.965-06:00 axis-b8a44f13e385 [ INFO    ] systemd[1]: Reloading.
2021-05-20T22:40:27.405-06:00 axis-b8a44f13e385 [ NOTICE  ] systemd[1]: /usr/lib/systemd/system/dbus.socket:5: ListenStream= references a path below legacy directory /var/run/, updating /var/run/dbus/system_bus_socket â†’ /run/dbus/system_bus_socket; please update the unit file accordingly.
2021-05-20T22:40:27.652-06:00 axis-b8a44f13e385 [ NOTICE  ] systemd[1]: /usr/lib/systemd/system/supervisedio_cgi.socket:6: ListenDatagram= references a path below legacy directory /var/run/, updating /var/run/iod/supervisediosocket â†’ /run/iod/supervisediosocket; please update the unit file accordingly.
2021-05-20T22:40:28.343-06:00 axis-b8a44f13e385 [ INFO    ] systemd[1]: Reloading.
2021-05-20T22:40:28.768-06:00 axis-b8a44f13e385 [ NOTICE  ] systemd[1]: /usr/lib/systemd/system/dbus.socket:5: ListenStream= references a path below legacy directory /var/run/, updating /var/run/dbus/system_bus_socket â†’ /run/dbus/system_bus_socket; please update the unit file accordingly.
2021-05-20T22:40:29.027-06:00 axis-b8a44f13e385 [ NOTICE  ] systemd[1]: /usr/lib/systemd/system/supervisedio_cgi.socket:6: ListenDatagram= references a path below legacy directory /var/run/, updating /var/run/iod/supervisediosocket â†’ /run/iod/supervisediosocket; please update the unit file accordingly.
2021-05-20T22:40:29.626-06:00 axis-b8a44f13e385 [ INFO    ] systemd[1]: Starting PoE power change notify...
2021-05-20T22:40:29.834-06:00 axis-b8a44f13e385 [ INFO    ] poe-housing-power-change[5784]: ()
2021-05-20T22:40:30.133-06:00 axis-b8a44f13e385 [ INFO    ] systemd[1]: poe-housing-power-change.service: Succeeded.
2021-05-20T22:40:30.135-06:00 axis-b8a44f13e385 [ INFO    ] systemd[1]: Started PoE power change notify.
2021-05-20T22:40:30.204-06:00 axis-b8a44f13e385 [ INFO    ] systemd[1]: usage-statistics.service: Succeeded.
2021-05-20T22:40:30.227-06:00 axis-b8a44f13e385 [ INFO    ] systemd[1]: Stopping object_detection...
2021-05-20T22:40:30.230-06:00 axis-b8a44f13e385 [ INFO    ] sdkobject_detection[5789]:  * Stopping object_detection...
2021-05-20T22:40:30.234-06:00 axis-b8a44f13e385 [ INFO    ] object_detection[0]: stopping object_detection
2021-05-20T22:40:30.261-06:00 axis-b8a44f13e385 [ INFO    ] sdkobject_detection[5789]: No /usr/local/packages/object_detection/object_detection found running; none killed.
2021-05-20T22:40:30.296-06:00 axis-b8a44f13e385 [ INFO    ] sdkobject_detection[5789]: [ ok ]
2021-05-20T22:40:30.298-06:00 axis-b8a44f13e385 [ INFO    ] systemd[1]: sdkobject_detection.service: Succeeded.
2021-05-20T22:40:30.299-06:00 axis-b8a44f13e385 [ INFO    ] systemd[1]: Stopped object_detection.
2021-05-20T22:40:30.320-06:00 axis-b8a44f13e385 [ INFO    ] sdkobject_detection[5796]:  * Starting object_detection...
2021-05-20T22:40:30.320-06:00 axis-b8a44f13e385 [ INFO    ] systemd[1]: Starting object_detection...
2021-05-20T22:40:30.390-06:00 axis-b8a44f13e385 [ INFO    ] object_detection[0]: starting object_detection
2021-05-20T22:40:30.399-06:00 axis-b8a44f13e385 [ INFO    ] sdkobject_detection[5796]: [ ok ]
2021-05-20T22:40:30.403-06:00 axis-b8a44f13e385 [ INFO    ] systemd[1]: Started object_detection.
2021-05-20T22:40:47.881-06:00 axis-b8a44f13e385 [ INFO    ] dbus-daemon[389]: [system] Activating via systemd: service name='com.axis.FirmwareManager1' unit='dbus-com.axis.FirmwareManager1.service' requested by ':1.2091' (uid=155 pid=5865 comm="gdbus call -y -d com.axis.FirmwareManager1 -o /com")
2021-05-20T22:40:47.897-06:00 axis-b8a44f13e385 [ INFO    ] systemd[1]: Starting Firmware Manager...
2021-05-20T22:40:47.907-06:00 axis-b8a44f13e385 [ INFO    ] dbus-daemon[389]: [system] Successfully activated service 'com.axis.FirmwareManager1'
2021-05-20T22:40:47.913-06:00 axis-b8a44f13e385 [ INFO    ] systemd[1]: Started Firmware Manager.
2021-05-20T22:40:47.963-06:00 axis-b8a44f13e385 [ INFO    ] dbus-daemon[389]: [system] Activating via systemd: service name='com.axis.UsageStatistics1' unit='dbus-com.axis.UsageStatistics1.service' requested by ':1.2093' (uid=155 pid=5876 comm="gdbus call -y -d com.axis.UsageStatistics1 -o /com")
2021-05-20T22:40:47.983-06:00 axis-b8a44f13e385 [ INFO    ] systemd[1]: Starting Usage statistics information...
2021-05-20T22:40:47.992-06:00 axis-b8a44f13e385 [ INFO    ] dbus-daemon[389]: [system] Successfully activated service 'com.axis.UsageStatistics1'
2021-05-20T22:40:47.996-06:00 axis-b8a44f13e385 [ INFO    ] systemd[1]: Started Usage statistics information.
2021-05-20T22:41:22.404-06:00 axis-b8a44f13e385 [ INFO    ] systemd[1]: fwmgr.service: Succeeded.
2021-05-20T22:41:45.388-06:00 axis-b8a44f13e385 [ INFO    ] dbus-daemon[389]: [system] Activating via systemd: service name='com.axis.FirmwareManager1' unit='dbus-com.axis.FirmwareManager1.service' requested by ':1.2170' (uid=155 pid=6364 comm="gdbus call -y -d com.axis.FirmwareManager1 -o /com")
2021-05-20T22:41:45.397-06:00 axis-b8a44f13e385 [ INFO    ] systemd[1]: Starting Firmware Manager...
2021-05-20T22:41:45.410-06:00 axis-b8a44f13e385 [ INFO    ] dbus-daemon[389]: [system] Successfully activated service 'com.axis.FirmwareManager1'
2021-05-20T22:41:45.414-06:00 axis-b8a44f13e385 [ INFO    ] systemd[1]: Started Firmware Manager.

joha-axis, 2021-05-21:
Currently, the examples are based on ACAP SDK 3.3, which requires a firmware version of 10.5 or later. Thus, you will have to update your firmware in order to run the example. I will make sure this is clarified in the repository.

You can check out the SDK to firmware compatibility schema here: https://www.axis.com/products/online-manual/s00004#t10164987

closed 2021-05-21 by joha-axis