AxisCommunications/acap3-examples #90: Web UI notification

ilsme, 2021-05-25:
Hi,

I'm wondering if there is an API to show the ACAP app's custom notifications for the user on the Web UI?

There are standard notifications like below but I didn't find a way to reuse them for custom notifications of my application.
[image]

Thanks.

mikaelli-axis, 2021-05-27:
This is not something we are currently supporting. The camera WebUI is somewhat limited in scope and focused around installation and initial configuration of the device. Commonly ACAPs include their own UI where these types of notifications can be displayed, alternatively rely on the event framework for sending notifications. Check the ACAP developer pages for more information on how to include configuration UI for your ACAP.

ilsme, 2021-06-02:
Got it, thanks.

closed 2021-06-02 by ilsme