AxisCommunications/acap3-examples #92: AxEvents > Companion App

keesschollaart81, 2021-06-02:
I see the Axis companion app can react to both VMD and Axis Object Analytics events. Is there a way for 3th party developers to raise their own events that can be consumed in this app? Would be great to utilize this push notification feature of it.

If not, can I push a event/message to either the VMD or the AOA topic?

mikaelli-axis, 2021-06-04:
You can define and publish events from your application using the axEvent API, but questions about what is possible with the Companion application is beyond what we can answer here. I would recommend contacting Axis customer support regarding how to configure the companion application to listen to custom application events.

keesschollaart81, 2021-06-04:
Will do! If I get a solution back, I'll report here.

closed 2021-06-04 by keesschollaart81