AxisCommunications/acap3-examples #94: Larod Converter > PyPi?

keesschollaart81, 2021-06-03:
Will the LarodConverter Python package every be published on PyPi? I'm converting my models as part of a CI/CD pipeline and it would be great to be able to convert TF Lite (etc) models to Larod without needing the entire Acap SDK!

As a workaround, I've extracted the whl file and I install it but it would be nice to have a centralized/versioned way to work with this package, much like all other packages do.

petterwa, 2021-06-04:
Larod service has added support for using tflite format directly which will be available from AxOS firmware 10.6. 
I.e. when using this firmware or later, the LarodConverter step can be skipped from your CI/CD pipeline and you can send in the .tflite model directly to larod.

keesschollaart81, 2021-06-04:
That's even better! Thanks!

closed 2021-06-04 by keesschollaart81