AxisCommunications/acap3-examples #102: Unable to build by following the instructions

jeremyVignelles, 2021-07-02:
**Issue description**
The various examples I tried (vdostream and axoverlay) couldn't be built by following exactly the documentation: the `docker build` steps fails, because `eap-create.sh` wants more info.
docker build is unable to be interactive, and the command stops.
See log below

Either the doc should recommend to create a package.conf file with the required options, or the build procedure should be modified completely.

**System setup**
Windows 10 21H1 / Docker for windows (Hyper-V) / git bash

**Logs**
```
 > [4/4] RUN . /opt/axis/acapsdk/environment-setup* && create-package.sh:
#7 1.730 make
#7 1.739 arm-linux-gnueabihf-gcc  -mthumb -mfpu=neon -mfloat-abi=hard -mcpu=cortex-a9 -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/opt/axis/acapsdk/sysroots/armv7hf -L /opt/axis/acapsdk/sysroots/armv7hf/usr/lib  -O2 -pipe -g -feliminate-unused-debug-types  -pthread -I/opt/axis/acapsdk/sysroots/armv7hf/usr/include/vdo -I/opt/axis/acapsdk/sysroots/armv7hf/usr/include/gio-unix-2.0 -I/opt/axis/acapsdk/sysroots/armv7hf/usr/include/glib-2.0 -I/opt/axis/acapsdk/sysroots/armv7hf/usr/lib/glib-2.0/include -Wall -Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed -Wl,-z,relro,-z,now vdoencodeclient.c  -lvdostream -lgio-2.0 -lgobject-2.0 -lglib-2.0 -o vdoencodeclient
#7 1.962
#7 1.962 eap-create.sh
#7 1.971
#7 1.971 No package.conf file foundCreated package.conf using default values... ok
#7 2.087 Reading local package.conf... ok
#7 2.089 Package architecture: armv7hf
#7 2.198 Validating Package config...
#7 2.207
#7 2.207 [Error] * PACKAGENAME cannot be empty
#7 2.208 We need to fix package.conf Please answer the following questions:
#7 2.208
#7 2.214 * PACKAGENAME cannot be empty
#7 2.219 PACKAGENAME: Package name or description []
#7 2.220 eap-create.sh failed. Please fix above errors, before you can create a package
```

petterwa, 2021-07-02:
This issue is related to Windows - I get same build errors when building from a fresh installation on my Windows machine.  On ubuntu the examples build is successful.

jeremyVignelles, 2021-07-02:
What could it be then ? I assumed that since the files were copied into the container, the host machine wouldn't matter?

Does your docker ask you those questions when you build on linux ? Is that an automatic detection that fails somehow ?

petterwa, 2021-07-02:
It looks like the script for generating a default package.conf fails, but it's interesting that this problem just happens when running Windows on the host. 
The fast workaround until we found the issue in the script is just to add a package.conf file in the app directory and then it builds fine also on windows.

jeremyVignelles, 2021-07-02:
I just tried to comment out the "RUN" line from the vdostream from the Dockerfile.
Then, I ran:
```
docker build --tag vdostream .
docker run --rm -it vdostream
```

Into the container, I ran this:

```
. /opt/axis/acapsdk/environment-setup*
create-package.sh
```

When the script is asking me to enter the details, I hit ctrl+C and inspected the folder with `ls -al`

```
total 72
drwxr-xr-x 1 root root  4096 Jul  2 14:29 .
drwxr-xr-x 1 root root  4096 Jun 18 18:44 ..
-rwxr-xr-x 1 root root 11556 Jul  2 08:56 LICENSE
-rwxr-xr-x 1 root root   421 Jul  2 08:56 Makefile
-rw-r--r-- 1 root root     0 Jul  2 14:29 package.conf     
-rwxr-xr-x 1 root root 33604 Jul  2 14:29 vdoencodeclient  
-rwxr-xr-x 1 root root  7927 Jul  2 08:56 vdoencodeclient.c
```

It looks like it can create the package.conf file.
I had a look at `/opt/axis/acapsdk/sysroots/x86_64-pokysdk-linux/usr/bin/create-package.sh`, which calls the eap-create.sh script, which contains this code:

```sh
        if [ -z "$APPNAME" ]; then
                # If we find exactly one executable, suggest that file as default.
                # Otherwise, empty default.
                for F in *; do
                        if [ ! -d $F ] ; then
                                if { [ -x $F ]; }; then
                                        if [ "$APPNAME" ]; then
                                                # Allow only one match
                                                APPNAME=
                                                break
                                        fi
                                        APPNAME=$F
                                fi
                        fi
                done
        fi
```

My guess would be that the fact that there is no unix permission on windows allows every file to be executable when copied into the container, and that fails the build.
Indeed, if I add this line after the `COPY` line into the Dockerfile, the image builds fine: 

```
RUN chmod -x /opt/app/*
```

Could this line be added into the Dockerfile for Windows users ?

mikaelli-axis, 2021-07-12:
Possibly, though the auto detection scheme used is perhaps not ideal for any platform, as it could be confused by an application including more than one binary as well as other similar scenarios. For now I think you will have to add it to your own Dockerfiles, or just avoid the auto detection by providing a manually written manifest/package.conf

closed 2021-08-25 by mikaelli-axis