AxisCommunications/acap3-examples #104: Object Detection - with MQTT

joakimroeed, 2021-07-07:
I want to use the object-detection code but with MQTT. 
I see you have a "subscribe-to-event" and "send-event" code. Can this be implemented with the object-detection?
Any ideas on how to do that?

mikaelli-axis, 2021-07-12:
The current event examples use the in-camera event system and not MQTT, but they can certainly be combined with the object-detection example. We have had many requests for an MQTT example so it is on our todo list, but for now if you want to do that I suggest you include something like the Paho MQTT library[1] in your application. 

[1]: https://github.com/eclipse/paho.mqtt.c

joakimroeed, 2021-07-14:
I tried doing something like this to get MQTT installed with the object-detection code but it doesn't seem to work. 
This is in the Dockerfile. I would appreciate any help.

`# Paho MQTT
WORKDIR /opt/build
RUN git clone https://github.com/eclipse/paho.mqtt.c.git
WORKDIR /opt/build/paho.mqtt.c
RUN make
RUN cp /opt/build/paho.mqtt.c/src/*.h include/
RUN cp /opt/build/paho.mqtt.c/src/*.c include/`

mikaelli-axis, 2021-07-14:
From a glance it seems you are building Paho but then copying just the source code and not the built binaries, but it is not clear exactly how the rest of the project is set-up. I would suggest you look closer at e.g. the vdo-larod example to see how the YUV-library is compiled and installed. Hope this helps, I cannot really offer in-depth Paho support as this is not part of the SDK. There is a more comprehensive example of how to include 3rd party libraries in the works, so keep your eyes out for that one.

joakimroeed, 2021-07-14:
Is it possible to format the object-detection code to tensorflow-to-larod?
What I mean is to run object-detection in a container similar to tensorflow-to-larod?
Help would be appreciated.

mikaelli-axis, 2021-07-14:
In both cases, the application is built using our containerised SDK and in the tensorflow-to-larod example this includes the training. In the end the applications are packaged as ACAP3 applications for deployment to the camera. Starting with ACAP 4, which is to be released later this year, we will support running containers directly on newer camera models, so if this is interesting to you keep an eye open for the release announcement.

joakimroeed, 2021-07-15:
I can't for the life of me figure out how to get MQTT installed into my docker container. 
Please help.

joakimroeed, 2021-07-19:
I got the libraries into the docker container but I'm struggling with compiling it. Not sure what to include in the Makefile.
Please help.

carlcn, 2021-07-29:
When you have MQTT installed in the docker container, the only thing you need to add to the Makefile is a reference to the selected MQTT library:
```sh
LDLIBS  += -lyuv -ljpeg -lpaho-mqtt3cs
```

closed 2021-08-17 by petterwa

adeniakusuma, 2022-02-15:
do you solve your problem to use the object-detection code but with MQTT? because now I have a task to send object detection results with MQTT, and I am stuck with it. Please share if you have already solved it, thank you very much.