AxisCommunications/acap3-examples #110: vdo-larod object coordinate

awais40, 2021-07-30:
As by given code

```
uint8_t maxProb = 0;
size_t maxIdx = 0;
uint8_t* outputPtr = (uint8_t*) larodOutputAddr;
for (size_t j = 0; j < args.outputBytes; j++) {
    if (outputPtr[j] > maxProb) {
        maxProb = outputPtr[j];
        maxIdx = j;
    }
}
if (labels) {
    if (maxIdx < numLabels) {
        syslog(LOG_INFO, "Top result: %s with score %.2f%%", labels[maxIdx],
               (float) maxProb / 2.5f);
    } else {
        syslog(LOG_INFO, "Top result: index %zu with score %.2f%% (index larger "
               "than num items in labels file)",
               maxIdx, (float) maxProb / 2.5f);
    }
} else {
    syslog(LOG_INFO, "Top result: index %zu with score %.2f%%", maxIdx,
           (float) maxProb / 2.5f);
}

```
how can we get the coordinates of detected object?

carlcn, 2021-07-30:
The vdo-larod example is using an object classification model that returns a toplist of classified objects. It does not return any coordinates. To get coordinates of detected objects you have to use an object detection model shown in example object-detection.

awais40, 2021-08-05:
I have tested object detection code, it is working fine as it is using ssd mobilenet model file, So i have retained the ssd mobilenet with custom dataset, converted file to tflite, detecting fine on ubuntu then quantized To int8, still detection fine but when it convert to edgetpu using command

edgetpu_compiler ssd_mobilenet_int8.tflite

it generate edgetpu file, when I replace it in the object_detection model file it stop detecting, application runs without any issues on camera but didn’t detect anything. 

Any suggestions? Where i am doing wrong?

carlcn, 2021-08-06:
Check that your retrained model has the same input and output structure as the original model, that the types, sizes and locations are the same. The Netron application can be used to view a model: https://github.com/lutzroeder/netron

awais40, 2021-08-19:
@carlcn  thanks for your help, it was the output quantization difference, I was making the output of quantization to int8 but it should be float16. I was able to run Yolo and other model on the camera as well, Getting 3 to 4fps, which works for me. Thanks

closed 2021-08-21 by awais40