AxisCommunications/acap3-examples #112: Technical Specifications of TPU

xsacha, 2021-08-11:
**Issue description**
Is there any where that describes the technical specifications of the Google TPU onboard these cameras?

From what I can tell these two cameras I had came with two different TPUs.
P3255-LVE says it has a Google 'Yorktown'
Q1615 Mk III says it has a Google 'Apex'

I can't find any information about this Yorktown and not sure of the capability differences of it. I have had to return the P3255-LVE now but would like to know of its capability relative to the other Google TPUs.

I thought this would be the right repository to put this in. If it's not, you can redirect me. Thanks.

**System setup**
- *AXIS product/device (e.g., Q1615 Mk III):* Q1615 Mk III, P3255-LVE
- *Device firmware version (e.g., 10.5):* 10.6
- *SDK version (e.g., ACAP SDK 3.3):* 3.3

petterwa, 2021-08-11:
Yorktown and Apex was used internally at Axis during development phase for different form factors of the TPU. 
P3255-LVE has PCIe module
Q1615 Mk has LGA module
These are the exact same chip, just different form factors. Therefore, from a users point of view, there should not be any differences.

xsacha, 2021-08-12:
Thanks @petterwa

closed 2021-08-12 by xsacha