AxisCommunications/acap3-examples #114: AxOverlay example not working

srefshauge, 2021-09-01:
**Issue description**
When running the AxOverlay example, I am expecting to visibly see overlays appearing but nothing appears. The logs show no errors when running and I have followed all steps in the example.

Possible issue: There is an existing text overlay by the camera saying there isn't enough power.
"Not enough power, product requires POE class 4 or higher"
I'm not sure if this affects my own overlays but hopefully it wouldn't. I am providing the camera with 60W (rather than 75W) but that's the best I can do.

**System setup**
- *AXIS product/device (e.g., Q1615 Mk III):* P3255-LVE
- *Device firmware version (e.g., 10.5):* 10.6
- *SDK version (e.g., ACAP SDK 3.3):* 3.4

**Logs**
2021-09-01T11:37:16.766+10:00 axis-b8a44f0e42b1 [ INFO    ] axoverlay[601]: Render callback for camera: 1
2021-09-01T11:37:16.766+10:00 axis-b8a44f0e42b1 [ INFO    ] axoverlay[601]: Render callback for overlay: 1920 x 1088
2021-09-01T11:37:16.766+10:00 axis-b8a44f0e42b1 [ INFO    ] axoverlay[601]: Render callback for stream: 1920 x 1080
2021-09-01T11:37:16.771+10:00 axis-b8a44f0e42b1 [ INFO    ] axoverlay[601]: Adjust callback for overlay: 1920 x 1080
2021-09-01T11:37:16.771+10:00 axis-b8a44f0e42b1 [ INFO    ] axoverlay[601]: Adjust callback for stream: 1920 x 1080
2021-09-01T11:37:16.788+10:00 axis-b8a44f0e42b1 [ INFO    ] axoverlay[601]: Render callback for camera: 1
2021-09-01T11:37:16.788+10:00 axis-b8a44f0e42b1 [ INFO    ] axoverlay[601]: Render callback for overlay: 1920 x 1088
2021-09-01T11:37:16.788+10:00 axis-b8a44f0e42b1 [ INFO    ] axoverlay[601]: Render callback for stream: 1920 x 1080
.... (Continuously)

srefshauge, 2021-09-01:
Update: I got an AXIS POE Injector that does deliver 75W and now the overlays work!

petterwa, 2021-09-02:
The device has to shut something down when there is not enough power and I don't think it's strange that this may affect an API. It should however be visible in the error handling from the api that AxOverlay in this case doesn't work unless you switch to another power supply.

xsacha, 2021-09-03:
I may be wrong but my understanding was that the extra power requirement was just for the heater (as it is an outdoor camera). The heater would never be used in my climate anyway. So just as an aside, it would be good to have an ability to disable the heater (I could not find any way) or turn on an indoor mode, lowering the power requirement.

Other than the overlays, all the functions of the camera appeared to be working.

If the power is used for another purpose, it'd be nice if the overlay mentioned which feature was no longer available. I have another camera (60W) that tells me the PTZ functions will no longer work when running @ 30W for instance.

It does make it more difficult to use the camera as 75W isn't common with switches. Certainly none at our offices!