AxisCommunications/acap3-examples #120: Could not allocate tensors

JesperHvilsted, 2021-09-10:
[image]

Hello! I have an issue with the tensorflow-to-larod example. 
When I run the converted_model_edgetpu.tflite file that is available in the example, it works fine, but if I take the converted_model.tflite from the example and compile it into an edgetpu and run it on the camera, I get the issue on the picture above. 
The edgetpu file in the example is 1.721 KB and the one I compiled is 1725 KB. 
I have tried to compile both inside and outside the container, but ends up wtih the same result.
I'm on a Windows computer and I use an AXIS Q1615 Mk III Network Camera.

joha-axis, 2021-09-10:
Hi,

Are you using the newest version of the edgetpu_compiler, i.e., v16? If that is the case, the problem is likely that it defaults to using the edgetpu runtime v14, which is not currently on the camera firmware. Therefore, you currently need to pass the `--min_runtime_version 13` flag when you compile your model. If that doesn't work, let me know.

Thanks for the report. I'll make sure a note on this is added to the documentation.

JesperHvilsted, 2021-09-10:
Hi, 
Yes that solved the problem! Thanks for fast respons

closed 2021-09-10 by JesperHvilsted