AxisCommunications/acap3-examples #128: Questions: Application Example

acschristoph, 2021-09-27:
Is it possible to create an example application boilerplate that covers most parts of an ACAP application in the real world?

Means:

1. Serving a Configuration Page ( and Handling parameter updating during Configuration Page/CGI) /HTTP API / Parameter API
3. Send Events (Event API)
4. Access video stream (Video capture API)

There are examples for each individual stuff but not an application boilerplate/example that covers all the parts that basically is needed in my opinion to write an ACAP application like VMD4/ObjectsAnalytics like AXIS.

thanks

petterwa, 2021-09-30:
Our strategy so far has been to have small examples that shows how to use on or two api's. It would be good with a more complete example to speed up development time. 
Labeling this as an enhancement proposal.

acschristoph, 2021-10-01:
Thank you, sounds good.

closed 2021-10-21 by pataxis