AxisCommunications/acap3-examples #138: axoverlay_redraw +  AXOVERLAY_COLORSPACE_ARGB32 slower than expected

eccrow, 2021-11-04:
**Issue description**
 axoverlay_redraw(&error) blocks for 80-100ms when the palette is set to AXOVERLAY_COLORSPACE_ARGB32, is that expected?. When the palette is set to AXOVERLAY_COLORSPACE_4BIT_PALETTE blocking is minimal.

**System setup**
- *AXIS product/device (e.g., Q1615 Mk III):*
- *Device firmware version (e.g., 10.6):*
- *SDK version (e.g., ACAP SDK 3.4):*

pataxis, 2021-11-11:
What system setup did you have, i.e. which SDK, product and firmware did you use?

EarthCam, 2021-11-15:
I am working on a Q1615 Mk III, running firmware 10.6 and  I am using ACAP 3.4.

axoverlay_redraw just seems sluggish when AXOVERLAY_COLORSPACE_ARGB32 is set.
Alternatively we would like to try out  the OPENGLES backend but there is no example.

pataxis, 2021-12-01:
The team has not been able to reproduce this on 10.8.

closed 2021-12-01 by pataxis