AxisCommunications/acap3-examples #168: Memory leak in vdostream example

SamYip627, 2022-05-30:
## Issue description

For the vdostream example from the link https://github.com/AxisCommunications/acap3-examples/tree/master/vdostream, we found that there is memory leak when I change the stream format to "jpeg".

## System setup

AXIS P3265-LV

Docker Information:
ARCH=aarch64
VERSION=3.5
UBUNTU_VERSION=20.04
REPO=axisecp
SDK=acap-sdk

## Logs
May 30 17:15:29 axis-b8a44f3b8339 vdoencodeclient[3972475]: frame = 56117, type = jpeg, size = 20286
May 30 17:15:29 axis-b8a44f3b8339 vdoencodeclient[3972475]: frame = 56118, type = jpeg, size = 20280
May 30 17:15:29 axis-b8a44f3b8339 vdoencodeclient[3972475]: frame = 56119, type = jpeg, size = 20265
May 30 17:15:29 axis-b8a44f3b8339 vdoencodeclient[3972475]: frame = 56120, type = jpeg, size = 20264
May 30 17:15:29 axis-b8a44f3b8339 vdoencodeclient[3972475]: frame = 56121, type = jpeg, size = 20242

pataxis, 2022-05-30:
Hi and thank you for reporting this issue. Some additional questions,

- Which AXIS OS version are you using?
- You don't see a memory leak in other streaming formats like h264?

We have a known memory leak in VDO between 10.10 and 10.11 SR2 that we are about to document. It doesn't sound like what you have found but might be good to be aware of.

WilfR, 2022-05-30:
Hi SamYip627, 
  Just out of curiosity, how are you establishing that a memory leak is taking place? Do you run the "free" command, then run the vdostream example application, then run the "free" command again and compare the value of the "used" output from before and after running the example application? Or something else? (I'm a noob in this development environment. Apologies if this is a noob question.)
Thanks!

SamYip627, 2022-05-31:
@pataxis 
The firmware version of the camera is 10.10.73.
I don't see memory leak with h264.

SamYip627, 2022-05-31:
@WilfR 
I use "top" command. And the VSZ usage of the application is keep increasing.

WilfR, 2022-05-31:
Much appreciated @SamYip627 ! Thank you!

pataxis, 2022-05-31:
@SamYip627 Thanks! Could you try out AXIS OS 10.11 SR2 with type jpeg?

SamYip627, 2022-05-31:
Where can I get the 10.11 SR2? 
https://www.axis.com/support/firmware doesn't provide that for P3265-LV.

pataxis, 2022-05-31:
Ohh, that's a pity. I checked and the product is unfortunately wrongly marked in release notes for this service release. There is a bug for this product and it will hopefully get a new release in AXIS OS 10.11 SR7, but that's still ongoing.

We will try to reproduce this issue and get back to you.

Meanwhile, if you like, you could test AXIS OS 10.9.

pataxis, 2022-06-03:
I have tested and can see the memory leak on 10.10.73 and that it's fixed in 10.11.65.
But again, unfortunately for this P3265-LVE there is no 10.11 service release (SR) yet.

I will keep this issue open and pinned until there is a new release for P3265-LVE.

Thanks again for reporting!

pataxis, 2022-06-07:
@SamYip627  There is now a new release 10.11.76 for P3265-LVE on https://www.axis.com/support/firmware, could you confirm that it solves your issue?

SamYip627, 2022-06-08:
It is solved. Thanks

closed 2022-06-08 by pataxis