AxisCommunications/acap3-examples #191: How to convert object detection tflite to ambarella bin and anchor bin

maazullah96, 2023-01-04:
I converted the YOLOv5.tflite file to the required Ambarella bin format for the CV25 architecture . I think I also need an anchors bin file for running the object detection application on the CV25. Can you suggest how I can generate these two files for object detection? Also, how can I visualize or test the bin file to see if it is able to perform object detection? I tried using AWS SageMaker to convert the file, but I am not sure how to run the file on an object detection app. Is there anything missing in the documentation, or can you provide some guidance on this?

pataxis, 2023-01-12:
HI @maazullah96 , thanks for your question on https://github.com/AxisCommunications/acap3-examples/tree/main/object-detection-cv25. We'll get back to you shortly.

maazullah96, 2023-01-18:
Thanks, @pataxis and @Corallo  for the help.
Can you please add or update the documentation of the conversion of the object detection model to the ambarella format as well? because I tried the steps from [tensorflow to lard app](https://github.com/AxisCommunications/acap3-examples/tree/main/tensorflow-to-larod-cv25#compiling-for-cv25-dlpu) but my file is outputting only 0 or nan value . I think I am doing something wrong can you guide  me about the conversion process 
I am using yolov5 medium file 

The commands I am using are 

`

# First Step to activate The Environment
source /usr/local/amba-cv-tools-basic.2.5.1.0.2008.ubuntu-18.04/env/cv25.env

/usr/local/amba-cv-tools-basic.2.5.1.0.2008.ubuntu-18.04/cv2/tv2/exe/gen_image_list.py \
-f images -o /home/maaz/Downloads/ambarella_new/img_list.txt -ns -e .jpg -c 0 -d 0,0 -r 256,256 \
-bf /home/maaz/Downloads/ambarella_new/dra_image_bin/ \
-bo /home/maaz/Downloads/ambarella_new/dra_image_bin/draw_bin_list.txt

/usr/local/amba-cv-tools-basic.2.5.1.0.2008.ubuntu-18.04/cv2/tv2/exe/graph_surgery.py tf -p /home/maaz/Downloads/ambarella_new/yolo5-custom-int8.tflite -o /home/maaz/Downloads/ambarella_new/yolo5-custom-int8_iq.pb -isrc 'i:serving_default_input_1|is:1,416,416,3' -t model/tf_upsample_1/resize/ResizeNearestNeighbor,model/tf_upsample/resize/ResizeNearestNeighbor -on "StatefulPartitionedCall"

/usr/local/amba-cv-tools-basic.2.5.1.0.2008.ubuntu-18.04/cv2/tv2/exe/tfparser.py -p "/home/maaz/Downloads/ambarella_new/yolo5-custom-int8_iq.pb" -o yolo5-custom-int8 -of ./yolo5-custom-int8 -d 4 -isrc 'i:serving_default_input_1|is:1,3,416,416'

vas -auto -show-progress /home/maaz/Downloads/ambarella_new/yolo5-custom-int8/yolo5-custom-int8.vas

/usr/local/amba-cv-tools-basic.2.5.1.0.2008.ubuntu-18.04/env/local/cavalry_gen/bin/cavalry_gen -d vas_output/ -f /home/maaz/Downloads/ambarella_new/yolo5-custom-int8.bin

`

Corallo, 2023-01-20:
Unfortunately, we can't provide detailed instructions on specific models.
It looks like you are running the gen_image_list using the wrong image size as an argument. You need to provide values to the -r that are equal to the input size of your model. 
To get further support for the model conversion to cv25 I'd recommend you to contact directly Ambarella.

closed 2023-01-23 by pataxis