AxisCommunications/acap3-examples #196: customized app is kill when changing image property source

vuhailongkl97, 2023-02-13:
## Issue description
currently when I change image source's properties as Rotation I got a warning  like this
```
Do you want to change the source rotation?
Check that your apps work properly after. If they don't, restart the camera.
```
I would know events which raised for the event to my application ?
[image]

Could you give me some advice to handle the events (if have) to my app can work properly.
## System setup

- *AXIS product/device (e.g., Q1615 Mk III): Q1615 Mk III
- *Device firmware version (e.g., 10.5):* 11.2
- *SDK version (e.g., ACAP SDK 3.3):*  build with dockerfile in this repo

## Logs
on the web
```
Do you want to change the source rotation?
Check that your apps work properly after. If they don't, restart the camera.
```

pataxis, 2023-02-13:
Hi @vuhailongkl97 , thanks for reaching out.

The normal use case for image rotation that I know of is to adapt the image output to how a camera is mounted. But after mounting the camera the position is fixed and the rotation don't change.

Most if not all applications and APIs should have been developed for a fix scene or image that's never rotated. If rotation setting is changed it would affect this basic precondition and I think that's what the warning is about.

For products with auto-rotation, the camera itself can adjust the image to how it's physically mounted.

vuhailongkl97, 2023-02-14:
@pataxis thank you.
That means if users want the application to work after the changed rotation, they must restart the axis device, and on the software side, I can't fix that. Right?

pataxis, 2023-02-14:
> Check that your apps work properly after. If they don't, restart the camera.

I guess it depends on the application, some may not be affected, some may work after restarting the application and some need a restart of the device itself. It could also depend on which product and firmware version.

closed 2023-02-15 by vuhailongkl97