AxisCommunications/acap3-examples #201: AXEvent : ax_event_handler_declare always returns the same declaration id

polodroid74, 2023-02-23:
## Issue description

When trying to declare multiple events, calls to ax_event_handler_declare() always returns the same declaration id.
Here is a code snippet to illustrate where the problem occurs : 

```
int tar::setupFloodStateDeclaration(std::unique_ptr<tar::EventData> &event_data){ //This function is called with several EventData structures
    AXEventKeyValueSet *key_value_set = ax_event_key_value_set_new();
    GError *error = nullptr;
    const std::string flood_str = "Flood"+std::to_string(event_data -> flood_level);
    gboolean state_value = false;
    ax_event_key_value_set_add_key_value(key_value_set,
                                         flood_str.c_str(),
                                         NULL,
                                         &state_value,
                                         AX_VALUE_TYPE_BOOL,
                                         &error);

    ax_event_key_value_set_mark_as_data(key_value_set,
                                        flood_str.c_str(),
                                        NULL,
                                        &error);
    //Declare event
    ax_event_handler_declare(event_data -> event_handler,
                             key_value_set,
                             false, /* Indicates a state full declaration. */
                             &(event_data -> event_id), <------------------------------ Here we will always have 1.
                             (AXDeclarationCompleteCallback)declarationComplete,
                             event_data.get(),
                             &error);
    ax_event_key_value_set_free(key_value_set);
```

## System setup

- *AXIS product/device : AXIS P-1378LE
- *Device firmware version : 10.12.135
- *SDK version : 3.5

pataxis, 2023-03-01:
Hi @polodroid74 , thanks for reaching out, we'll get back to you when we had time to look in to this.

pataxis, 2023-03-03:
A quick look in the code, the id is expected to be stepped.

It's important to create only one `AxEventHandler` and reuse it throughout the program, see for example [subscribe_to_events](https://github.com/AxisCommunications/acap3-examples/blob/main/axevent/subscribe_to_events/app/subscribe_to_events.c#L452). The same should go for declaring events.
Could this be it?

polodroid74, 2023-03-06:
Yes this is it. Thanks a lot !

closed 2023-03-07 by pataxis