AxisCommunications/acap3-examples #220: axoverlay  example not working

pivpiv, 2024-02-26:
## Issue description

 When attempting to install the axoverlay example, the application installation process stops immediately.

## System setup

- *AXIS product/device : P1377-LE*
- *Device firmware version  :11.8.61*
- *SDK version: 3.5*

## Logs

2024-02-26T14:13:47.917+00:00 axis-b8a44f801096 [ INFO    ] axoverlay[2522]: Adjust callback for overlay: 2592 x 1944
2024-02-26T14:13:47.917+00:00 axis-b8a44f801096 [ INFO    ] axoverlay[2522]: Adjust callback for stream: 2592 x 1944
2024-02-26T14:13:47.917+00:00 axis-b8a44f801096 [ INFO    ] axoverlay[2522]: Max resolution (width x height): 2592 x 1944
2024-02-26T14:13:47.917+00:00 axis-b8a44f801096 [ INFO    ] axoverlay[2522]: Render callback for camera: 1
2024-02-26T14:13:47.917+00:00 axis-b8a44f801096 [ INFO    ] axoverlay[2522]: Render callback for overlay: 2592 x 1944
2024-02-26T14:13:47.917+00:00 axis-b8a44f801096 [ INFO    ] axoverlay[2522]: Render callback for stream: 2592 x 1944
2024-02-26T14:13:47.917+00:00 axis-b8a44f801096 [ WARNING ] axoverlay[2522]: g_error_free: assertion 'error != NULL' failed
2024-02-26T14:13:48.284+00:00 axis-b8a44f801096 [ INFO    ] axoverlay[2522]: Adjust callback for overlay: 2592 x 1944
2024-02-26T14:13:48.284+00:00 axis-b8a44f801096 [ INFO    ] axoverlay[2522]: Adjust callback for stream: 2592 x 1944
2024-02-26T14:13:48.892+00:00 axis-b8a44f801096 [ WARNING ] axoverlay[1170]: Last message 'g_error_free: assert' repeated 1 times, suppressed by syslog-ng on axis-b8a44f801096
2024-02-26T14:13:48.998+00:00 axis-b8a44f801096 [ INFO    ] axoverlay[2522]: Fontconfig error: No writable cache directories
2024-02-26T14:13:48.998+00:00 axis-b8a44f801096 [ INFO    ] axoverlay[2522]: Render callback for camera: 1
2024-02-26T14:13:48.998+00:00 axis-b8a44f801096 [ INFO    ] axoverlay[2522]: Render callback for overlay: 2592 x 1944
2024-02-26T14:13:48.998+00:00 axis-b8a44f801096 [ INFO    ] axoverlay[2522]: Render callback for stream: 2592 x 1944
2024-02-26T14:13:49.284+00:00 axis-b8a44f801096 [ INFO    ] axoverlay[1170]: Last message 'Fontconfig error: No' repeated 3 times, suppressed by syslog-ng on axis-b8a44f801096
2024-02-26T14:13:49.284+00:00 axis-b8a44f801096 [ INFO    ] axoverlay[2522]: Adjust callback for overlay: 2592 x 1944
2024-02-26T14:13:49.284+00:00 axis-b8a44f801096 [ INFO    ] axoverlay[2522]: Adjust callback for stream: 2592 x 1944
2024-02-26T14:13:49.862+00:00 axis-b8a44f801096 [ ERR     ] axoverlay[2522]: Failed to draw overlays: GDBus.Error:com.axis.Overlay2.Error.Internal: Internal error
2024-02-26T14:13:49.862+00:00 axis-b8a44f801096 [ INFO    ] axoverlay[2522]: Adjust callback for overlay: 2592 x 1944
2024-02-26T14:13:49.862+00:00 axis-b8a44f801096 [ INFO    ] axoverlay[2522]: Adjust callback for overlay: 2592 x 1944
2024-02-26T14:13:49.862+00:00 axis-b8a44f801096 [ INFO    ] axoverlay[2522]: Adjust callback for stream: 2592 x 1944
2024-02-26T14:13:49.862+00:00 axis-b8a44f801096 [ INFO    ] axoverlay[2522]: Adjust callback for stream: 2592 x 1944
2024-02-26T14:13:49.862+00:00 axis-b8a44f801096 [ INFO    ] axoverlay[2522]: Render callback for camera: 1
2024-02-26T14:13:49.862+00:00 axis-b8a44f801096 [ INFO    ] axoverlay[2522]: Render callback for overlay: 2592 x 1944
2024-02-26T14:13:49.862+00:00 axis-b8a44f801096 [ INFO    ] axoverlay[2522]: Render callback for stream: 2592 x 1944
2024-02-26T14:13:49.935+00:00 axis-b8a44f801096 [ INFO    ] axoverlay[2522]: Render callback for camera: 1
2024-02-26T14:13:49.935+00:00 axis-b8a44f801096 [ INFO    ] axoverlay[2522]: Render callback for overlay: 2592 x 1944
2024-02-26T14:13:49.935+00:00 axis-b8a44f801096 [ INFO    ] axoverlay[2522]: Render callback for stream: 2592 x 1944
2024-02-26T14:13:50.127+00:00 axis-b8a44f801096 [ INFO    ] axoverlay[2522]: Adjust callback for overlay: 2592 x 1944
2024-02-26T14:13:50.127+00:00 axis-b8a44f801096 [ INFO    ] axoverlay[2522]: Adjust callback for stream: 2592 x 1944
2024-02-26T14:13:50.721+00:00 axis-b8a44f801096 [ WARNING ] axoverlay[2522]: g_error_free: assertion 'error != NULL' failed

Additionally, the application cannot be installed on firmware  10.12.221

pivpiv, 2024-03-19:
Any suggestions on how to use Axoverlay on a P1377-LE ?

vivekatoffice, 2024-03-19:
Hi @pivpiv ,
I can reproduce the error when I set the capture rate to "5MP 4:3 @ 25/30 fps (WDR)":

```syslog
2024-02-26T14:13:47.917+00:00 axis-b8a44f801096 [ INFO ] axoverlay[2522]: Adjust callback for overlay: 2592 x 1944
```

# Alternative which can help you:
Able to run the application with the following settings and Axis OS:
[image]

[image]

[image]