AxisCommunications/acap3-examples #221: Issue with compatibility and shared libs, axovleray example with firmware 10.12.228

Cacsjep, 2024-03-26:
Hey, 

I look into the compatibility tables, and maybe i missunderstand somehting.
With my app I had an issue when i try to start it. To get sure its not an error with my docker or manifest etc., 
I use the axoverlay example without any modification to reproduce it:

`2024-03-26T07:24:33.193+01:00 axis-b8a44f504193 [ INFO    ] axoverlay[839099]: /usr/local/packages/axoverlay/axoverlay: error while loading shared libraries: libaxoverlay.so.1: cannot open shared object file: No such file or directory`

 - M3206-LVE
 - 10.12.228
 - SDK 3.5

Based on sdk compatiblity acap-sdk 3.5 should work with my firmware beacuse its higher as 10.9 
[image]

When i search on camera via ssh, i just can find 
`/usr/lib/libvideooverlay.so`
but no axoverlay.so 

```
-rwxr-xr-x    1 root     root       17.8K Apr  6  2011 libax_daemon.so
lrwxrwxrwx    1 root     root          17 Apr  6  2011 libaxaudio.so -> libaxaudio.so.1.0
lrwxrwxrwx    1 root     root          17 Apr  6  2011 libaxaudio.so.1 -> libaxaudio.so.1.0
-rwxr-xr-x    1 root     root       21.9K Apr  6  2011 libaxaudio.so.1.0
lrwxrwxrwx    1 root     root          16 Apr  6  2011 libaxcert.so -> libaxcert.so.1.0
lrwxrwxrwx    1 root     root          16 Apr  6  2011 libaxcert.so.1 -> libaxcert.so.1.0
-rwxr-xr-x    1 root     root       13.8K Apr  6  2011 libaxcert.so.1.0
-rwxr-xr-x    1 root     root       13.9K Apr  6  2011 libaxcgijson.so
-rw-r--r--    1 root     root      181.2K Apr  6  2011 libaxe.so
lrwxrwxrwx    1 root     root          17 Apr  6  2011 libaxevent.so -> libaxevent.so.1.1
lrwxrwxrwx    1 root     root          17 Apr  6  2011 libaxevent.so.1 -> libaxevent.so.1.1
-rwxr-xr-x    1 root     root       49.8K Apr  6  2011 libaxevent.so.1.1
lrwxrwxrwx    1 root     root          16 Apr  6  2011 libaxhttp.so -> libaxhttp.so.1.0
lrwxrwxrwx    1 root     root          16 Apr  6  2011 libaxhttp.so.1 -> libaxhttp.so.1.0
-rwxr-xr-x    1 root     root       17.8K Apr  6  2011 libaxhttp.so.1.0
-rwxr-xr-x    1 root     root       13.8K Apr  6  2011 libaxlog.so
lrwxrwxrwx    1 root     root          19 Apr  6  2011 libaxpackage.so -> libaxpackage.so.1.0
lrwxrwxrwx    1 root     root          19 Apr  6  2011 libaxpackage.so.1 -> libaxpackage.so.1.0
-rwxr-xr-x    1 root     root        9.8K Apr  6  2011 libaxpackage.so.1.0
lrwxrwxrwx    1 root     root          21 Apr  6  2011 libaxparameter.so -> libaxparameter.so.1.0
lrwxrwxrwx    1 root     root          21 Apr  6  2011 libaxparameter.so.1 -> libaxparameter.so.1.0
-rwxr-xr-x    1 root     root       25.9K Apr  6  2011 libaxparameter.so.1.0
lrwxrwxrwx    1 root     root          15 Apr  6  2011 libaxptz.so -> libaxptz.so.1.0
lrwxrwxrwx    1 root     root          15 Apr  6  2011 libaxptz.so.1 -> libaxptz.so.1.0
-rwxr-xr-x    1 root     root       37.9K Apr  6  2011 libaxptz.so.1.0
lrwxrwxrwx    1 root     root          19 Apr  6  2011 libaxstorage.so -> libaxstorage.so.1.0
lrwxrwxrwx    1 root     root          19 Apr  6  2011 libaxstorage.so.1 -> libaxstorage.so.1.0
-rwxr-xr-x    1 root     root       13.8K Apr  6  2011 libaxstorage.so.1.0
lrwxrwxrwx    1 root     root          16 Apr  6  2011 libaxvx.so.0 -> libaxvx.so.0.3.5
-rwxr-xr-x    1 root     root       98.1K Apr  6  2011 libaxvx.so.0.3.5
```

Is axoverlay the only axlib that is not usable !? or why is this missing ?

thanks

losamtz, 2024-03-26:
Hi @Cacsjep ,

I guess your camera is a Ambarella S5L & the API supports products with the following chips:

    ARTPEC-8
    ARTPEC-7
    ARTPEC-6

https://axiscommunications.github.io/acap-documentation/docs/acap-sdk-version-3/api/#overlay-api

Regards

Cacsjep, 2024-03-26:
Thank u for clearification, its true its S5L,

closed 2024-03-26 by Cacsjep

kendsouza-sice, 2025-04-24:
Gday

In relation to this thread.
I wanted to ask if there are any alternate solutions to create overlays on devices that are running older chipsets? 
I noticed the unit that I'm working with (Axis Q6125-LE 9.80.85 Ambarella S3L) supports Cairo.
Please let me know if it's possible. Thanks

- Ambarella CV25
- Ambarella S5L
- Ambarella S5
- Ambarella S3L
- Ambarella S2L