AxisCommunications/acap3-examples #228: Can't make docker image

Ajiro3, 2024-12-19:
## Issue description
I tried to build the image as shown in the Quickstart.
docker build --build-arg ARCH=aarch64 --build-arg CHIP=artpec8 --tag obj_detect .
It shows an error at the step of COPY yuv/*.patch /opt/build/libyuv.
 
## System setup
- *AXIS product/device (e.g., Q1615 Mk III): P1467-LE*
- *Device firmware version (e.g., 10.5): 11.3.70*
- *SDK version (e.g., ACAP SDK 3.3): *

## Logs

Log excerpt from the affected system. For AXIS devices, SSH:ing to the device and running `journalctl -f` provides a running system log which can be used to extract a relevant section of the log.
[image]

WilfR, 2024-12-20:
Hi Ajiro, 
  I'm not an Axis employee but I have built a few ACAPs. I will try to build the same Docker image and see if I get the same results. However I am not sure what specific Docker image you are referring to when you write "as shown in the Quickstart". Do you have a link to the specific Github repository that contains the project that you are trying to build? 
Thanks, 
Wilf.

WilfR, 2024-12-20:
D'Oh! It must be the "object-detection" project in the acap3-examples repo, right? OK, I will try to build that guy.

WilfR, 2024-12-20:
Well maybe I was mistaken about which project you are trying to build. If I clone the "acap3-examples" repo to my computer, cd to the acap3-examples/object-detection directory, and execute the command "docker build --build-arg ARCH=aarch64 --build-arg CHIP=artpec8 --tag obj_detect ." my build runs successfully to completion. The most likely explanation is that I have built the wrong thing. Can you confirm what specific project from which repo you are trying to build?

Ajiro3, 2024-12-20:
Hi WilfR,
Thank you for your response.
My concern is acap3-examples/object-detection folder.
I use Docker Desktop for Windows. Are you using Linux?
Here is my Dokerfile.
# object-detection
ARG ARCH=armv7hf
ARG VERSION=3.5
ARG UBUNTU_VERSION=20.04
ARG REPO=axisecp
ARG SDK=acap-sdk

FROM ${REPO}/${SDK}:${VERSION}-${ARCH}-ubuntu${UBUNTU_VERSION}

# Build libyuv
WORKDIR /opt/build
# TODO: Investigate why server certs can't be verified
RUN GIT_SSL_NO_VERIFY=1 git clone -n https://chromium.googlesource.com/libyuv/libyuv

WORKDIR /opt/build/libyuv
ARG libyuv_version=5b6042fa0d211ebbd8b477c7f3855977c7973048
RUN git checkout ${libyuv_version}

# ** modify not to get libyuv patch

# COPY yuv/*.patch /opt/build/libyuv

ARG ARCH=armv7hf
RUN if [ "$ARCH" = armv7hf ]; then \
  # git apply ./*.patch && \
  CXXFLAGS=' -O2 -mthumb -mfpu=neon -mfloat-abi=hard -mcpu=cortex-a9 -fomit-frame-pointer' \
  make -j -f linux.mk CXX=arm-linux-gnueabihf-g++ CC=arm-linux-gnueabihf-gcc && \
  arm-linux-gnueabihf-strip --strip-unneeded libyuv.so* ; \
  elif [ "$ARCH" = aarch64 ]; then \
  # git apply ./*.patch && \
  make -j -f linux.mk CXX=/usr/bin/aarch64-linux-gnu-g++ CC=/usr/bin/aarch64-linux-gnu-gcc && \
  aarch64-linux-gnu-strip --strip-unneeded libyuv.so* ; \
  else \
  printf "Error: '%s' is not a valid value for the ARCH variable\n", "$ARCH"; \
  exit 1; \
  fi

# Build libjpeg-turbo
WORKDIR /opt/build
RUN apt-get update && apt-get install --no-install-recommends -y cmake && \
  apt-get clean && \
  rm -rf /var/lib/apt/lists/*
RUN git clone --branch 2.0.6 https://github.com/libjpeg-turbo/libjpeg-turbo.git

WORKDIR /opt/build/libjpeg-turbo/build
RUN if [ "$ARCH" = armv7hf ]; then \
  gCFLAGS=' -O2 -mthumb -mfpu=neon -mfloat-abi=hard -mcpu=cortex-a9 -fomit-frame-pointer' \
  CC=arm-linux-gnueabihf-gcc cmake -G"Unix Makefiles" .. && \
  make -j; \
  elif [ "$ARCH" = aarch64 ]; then \
  CC=/usr/bin/aarch64-linux-gnu-gcc cmake -G"Unix Makefiles" .. && \
  make ; \
  else \
  printf "Error: '%s' is not a valid value for the ARCH variable\n", "$ARCH"; \
  exit 1; \
  fi

# Copy the built libraries to application folder
WORKDIR /opt/app
COPY ./app /opt/app/
ARG BUILDDIR=/opt/build/libyuv
RUN mkdir -p lib include && \
  cp /opt/build/libjpeg-turbo/build/*.so* lib/ && \
  cp /opt/build/libjpeg-turbo/build/*.h include/ && \
  cp /opt/build/libjpeg-turbo/*.h include/ && \
  cp ${BUILDDIR}/libyuv.so* lib/ && \
  cp -a ${BUILDDIR}/include/. include && \
  ln -s libyuv.so.1 lib/libyuv.so && \
  ln -s libyuv.so.1 lib/libyuv.so.1.0

# Get pretrained models
ARG CHIP=
# RUN if [ "$CHIP" = cpu ] || [ "$CHIP" = artpec8 ]; then \
#         curl -L -o converted_model.tflite \
#             https://github.com/google-coral/test_data/raw/master/ssd_mobilenet_v2_coco_quant_postprocess.tflite ; \
#     elif [ "$CHIP" = edgetpu ]; then \
#         curl -L -o converted_model.tflite \
#             https://github.com/google-coral/test_data/raw/master/ssd_mobilenet_v2_coco_quant_postprocess_edgetpu.tflite ; \
#     else \
#         printf "Error: '%s' is not a valid value for the CHIP variable\n", "$CHIP"; \
#         exit 1; \
#     fi

# Download the labels
# RUN mkdir -p label && \
#     curl -L -o label/labels.txt https://github.com/google-coral/test_data/raw/master/coco_labels.txt

# # Setup the model directory
# RUN mkdir -p model && \
#     cp converted_model.tflite model/converted_model.tflite
COPY detect.tflite detect.tflite
COPY parking.txt parking.txt
RUN mkdir -p label && \
  cp parking.txt label/labels.txt
# Setup the model directory
RUN mkdir -p model && \
  cp detect.tflite model/converted_model.tflite
# Building the ACAP application
RUN cp /opt/app/manifest.json.${CHIP} /opt/app/manifest.json && \
  . /opt/axis/acapsdk/environment-setup* && acap-build . \
  -a 'label/labels.txt' -a 'model/converted_model.tflite'

WilfR, 2024-12-20:
Hi Ajiro, 
  OK, thanks for posting your Dockerfile - that was a good idea! Here is the short answer: I believe that you are using an old and obsolete version of the acap3-examples repository. You should get the latest version of the acap3-examples repository (or update your local version) and then try to rebuild the object-detection project using the latest code. 

  I think that in April 2023 the Dockerfile for object-detection was modified to remove any dependency on the YUV library. You can see the changes made to the Dockerfile in this commit: https://github.com/AxisCommunications/acap3-examples/commit/fcc55278a486f1ef775f9060a101badc3b04ed40
  With this latest version of the acap3-examples repository I can build the object-detection project using the "docker build --build-arg ARCH=aarch64 --build-arg CHIP=artpec8 --tag obj_detect ." command that you provided. The build produces some scary-looking warning messages near the end of its run when it is compiling object_detection.c, but they are warnings only, not errors, and in this demo project these warnings can be ignored. 
  I was building the object-detection project on my Windows computer, just from a "cmd.exe" command window. 
Good luck!
Cheers, 
Wilf.

Ajiro3, 2024-12-21:
Hi WilfR-san,
Thank you for your kind advice.
I got its latest version and succeeded to build its .eap file.
And it looks like to work properly on my Cam.
Thank you!
Masataka

WilfR, 2024-12-21:
That is great news - thanks for letting me know! I am very happy to hear that your problem is resolved and I wish you good luck with your future ACAP projects!
Cheers, 
Wilf.

closed 2025-01-17 by pataxis

Ajiro3, 2025-01-17:
Hi WilfR,
Do you have any experiance that the Docker Daemon on Cam dose not respond with the mouted SD Card? 
It has been working fine with SD card. But suddenly it does not respond. When I unmount SD card, it works fine as before.
There are some recorded vidos on SD card and I can save video on it. So SD card might be OK. 
But Docker daemon might fail to access to SD card.
Regards,
Masataka

WilfR, 2025-01-20:
Hi Masataka, 
  I am sorry - I have no experience using Docker Daemon on the cameras, or for that matter, writing to the SD card. I'm afraid that I have no good suggestions for you. (Perhaps try restarting the camera??)
Cheers, 
Wilf.